opentrack/opentrack #416: Toggle calibration error

olegiy, 2016-08-15:
invert values
https://youtu.be/ZTR1XrmKPjw

sthalik, 2016-08-15:
It's not an error. It uses a statistical method. There's a link how to make the method detect the center properly in this dialog, in blue.

sthalik, 2016-08-15:
I'm sorry you are right.

sthalik, 2016-08-15:
@olegiy since when was it broken?

olegiy, 2016-08-15:
I discovered today. In previous versions did not check. ((

sthalik, 2016-08-15:
@olegiy build for you https://www.dropbox.com/s/vs2k93qlsg7q38s/opentrack-pt-tcalib-20160815.7z

Please report crashes since I changed the compiler from GNU 5.4.1 to 6.1.1 again.

olegiy, 2016-08-15:
https://www.dropbox.com --> "This file is no longer here"

Len62, 2016-08-17:
I'm getting the same issue in RC11. If I manually invert the numbers all is fine. As olegiy stated the test version is no longer there. Only happens when performing a new calibration. Previously saved ini files work fine.

sthalik, 2016-09-04:
Test build(s): https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

Len62, 2016-09-05:
No issues with calibration in this build.

sthalik, 2016-09-05:
Thank you for the report.

closed 2016-09-05 by sthalik