opentrack/opentrack #418: Camera offset, unable to get accurate game data readings

Spectator6, 2016-08-16:
Hello hello!  Once again, great app, thanks for creating and supporting so actively!

I had camera offset issues in earlier builds and I'm affecting by them again in rc49p10. After struggling to correct this issue to no avail, I thought I'd check in here in case a) I'm an idiot and not doing something right, which is entirely possible or b) it's an actual issue with how the points are being tracked or c) some mixture of both.

[Picture of offset values manually zeroed and camera location](http://imgur.com/a/z3DG3)

[gfycat of straight lateral head turns with settings as shown in picture](https://gfycat.com/FlickeringNauticalDutchsmoushond)

Something I've noticed is that even with the roll output disabled in settings, the headmotions are still translated in game data with roll, though isolated roll only inputs are not registered. If I enable roll output, the output is simply more pronounced from that shown and isolated roll inputs are interpreted.

Also, any adjustments I make to the TX, TY, and TZ values do not affect the game data readings (ie I cannot get them to zero out, they stay as they are shown). Not sure if this is required at all in the calibration, they do properly zero out when using the zero command in the program.

Thank you again! I'll try and share whatever I can to help improve the software! - All the best

sthalik, 2016-08-16:
Hey,

Camera offset is meant to be set to camera skew to keep axis interconnect low. On the other hand you have "center pose offset" that merely adjusts the pose by a certain value without affecting how continuous motion works.

If you rotated roll by 180, pitch and yaw would swap signs. If you rotated roll by 90, pitch and yaw would swap themselves.

Roll output will be indeed affected since the pose is pre-rotated and rotations don't come straight anymore. 

I recommend you set camera offset to the exact rotation raw data when you look straight at the monitor. Or don't set it if the skew isn't big.

Spectator6, 2016-08-16:
Thank you for the fast reply!

To make sure I'm understanding you correctly, I need to set the Yaw, Pitch, and Roll values to 12, 21, -24 so they match what's listed on "Raw tracker data". Then I should hit the shortcut key to "Center" the view, which should cause the "Game data" values for Yaw, Pitch, and Roll to zero. 

Did I get that right?

sthalik, 2016-08-17:
Yes. Then you should get less interconnect when moving around.

Spectator6, 2016-08-17:
Thanks for help and your patience! It's tracking head turns *MUCH* more accurately now that I've matched the raw data values in the camera offset, as you suggested. Thank you again for the great support!

sthalik, 2016-08-17:
No worries.

I'm adding an experimental centering method for cameras that, if works, should eliminate the need for an offset. But this needs wider testing first.

closed 2016-08-17 by sthalik