opentrack/opentrack #419: Can't configure camera options for PsEye

Creat, 2016-08-18:
Hello,

another issue which may or may not be related to OpenTrack. I only just got a Playstation Eye, primarily for the frame rate. Previously I was using a Logitech C920.

With the Playstation Eye selected in the PointTracker Settings, when I click on "Open" for the Camera settings dialog, all controls in that window are disabled. It does work in other programs, specifically the "CL-Eye Test"-App that's part of the CL-Driver package (afaik that's the only real driver for windows) can open the settings just fine and the controls work. It doesn't seem to make a difference if the camera is active or not.

The controls for the C920 do work when opened from the same PT-Dialog though, hence my not knowing if this is even OpenTracks fault... So far I haven't turned up anything with googling for the issue though...

Thanks
Creat

Len62, 2016-08-18:
I think this may still be a work in progress. Originally, IIRC, open camera settings would not even take you to the Cleye settings. Typically I find that once set I do not need to modify the settings so using the cleye settings works fine. Although I guess it would be nice to access them directly from OT.

sthalik, 2016-08-20:
It's a problem in the library we're using for camera input. I'm willing to accept patches to fix that - known - issue. However doing it all right requires a few days of research by itself, and I'd rather remove the "camera options" dialog than spend that time.

sthalik, 2016-08-20:
WONTFIX, patches welcome.

closed 2016-08-20 by sthalik

Creat, 2016-08-20:
That's fine, since it can easily be configured when opened from any other camera app.
Please don't remove the actual button though, since it does work fine with some cameras as mentioned above.