opentrack/opentrack #422: Toggle while held

individ2016, 2016-08-22:
I've found that "Toggle while held" function does not work in RC11.. Or i dont understand how it does.. I was binded same button on joystick on center and "Toggle while held" functions and opentrack returns center view, but doesnt work then, before i click stop and start buttons again..

individ2016, 2016-08-22:
And thank you a lot for this wanderfull soft!!! :)

MathijsG, 2016-08-31:
You added the same button to different settings? Of course that will conflict then.

MathijsG, 2016-08-31:
Hmm, I do encounter a bug there. Sometimes randomly the tracking also stops while just once pressing that certain button. And only by having that certain button pressed for a longer time, will let it be track again.

sthalik, 2016-08-31:
Is your CPU usage 100% or very high at the time?

sthalik, 2016-08-31:
Or swap file enabled with swapping occuring, something like that signifying high load.

sthalik, 2016-08-31:
I don't see the bug. Please disregard earlier comments.

MathijsG, 2016-08-31:
No, just about 8%. Trying it without gaming, just for bug squatting. It happens immediately, I assigned 'W' to toggle while held, and by pressing it a brief moment it immediately stops with tracking even when the button isn't pressed.

My profile for this:

`[Curves-tx]
point-count=1
point-0-x=60
point-0-y=100

[opentrack-ui]
camera-pitch=0
camera-roll=0
camera-yaw=0
center-at-startup=true
compensate-translation=true
compensate-translation-disable-z-axis=false
filter-dll=Accela
pitch-alt-axis-sign=true
pitch-invert-sign=false
pitch-source-index=4
pitch-zero-pos=0
protocol-dll=freetrack 2.0 Enhanced
roll-alt-axis-sign=false
roll-invert-sign=false
roll-source-index=5
roll-zero-pos=0
tracker-dll=PointTracker 1.1
use-system-tray=true
x-alt-axis-sign=false
x-invert-sign=false
x-source-index=0
x-zero-pos=0
y-alt-axis-sign=false
y-invert-sign=false
y-source-index=1
y-zero-pos=0
yaw-alt-axis-sign=false
yaw-invert-sign=false
yaw-source-index=3
yaw-zero-pos=0
z-alt-axis-sign=false
z-invert-sign=false
z-source-index=2
z-zero-pos=0
button-center=-1
button-start-tracking=-1
button-stop-tracking=-1
button-toggle=-1
button-toggle-tracking=-1
button-zero=-1
centering-method=1
guid-center=
guid-start-tracking=
guid-stop-tracking=
guid-toggle=
guid-toggle-tracking=
guid-zero=
keycode-center=Home
keycode-start-tracking=
keycode-stop-tracking=
keycode-toggle=
keycode-toggle-tracking=End
keycode-zero=
button-restart-tracking=-1
button-toggle-press=-1
button-zero-press=-1
guid-restart-tracking=
guid-toggle-press=
guid-zero-press=
keycode-restart-tracking=
keycode-toggle-press=W
keycode-zero-press=
tracklogging-enabled=false
tracklogging-filename=
use-camera-offset-from-centering=false

[Curves-tx_alt]
point-count=1
point-0-x=100
point-0-y=100

[Curves-ty]
point-count=1
point-0-x=60
point-0-y=100

[Curves-ty_alt]
point-count=1
point-0-x=100
point-0-y=100

[Curves-tz]
point-count=1
point-0-x=24
point-0-y=100

[Curves-tz_alt]
point-count=1
point-0-x=100
point-0-y=100

[Curves-rx]
point-count=1
point-0-x=30
point-0-y=180

[Curves-rx_alt]
point-count=2
point-0-x=20
point-0-y=90
point-1-x=30
point-1-y=180

[Curves-ry]
point-count=1
point-0-x=20
point-0-y=180

[Curves-ry_alt]
point-count=1
point-0-x=24
point-0-y=180

[Curves-rz]
point-count=3
point-0-x=0
point-0-y=0
point-1-x=2
point-1-y=0
point-2-x=30
point-2-y=45

[Curves-rz_alt]
point-count=1
point-0-x=180
point-0-y=180

[tracker-pt]
active-model-panel=1
camera-fov=56
camera-fps=120
camera-name=PS3Eye Camera
camera-res-height=480
camera-res-width=640
cap-x=40
cap-y=60
cap-z=100
clip-by=70
clip-bz=80
clip-ty=40
clip-tz=30
dynamic-pose-resolution=true
init-phase-timeout=2000
m_01-x=0
m_01-y=0
m_01-z=0
m_02-x=0
m_02-y=0
m_02-z=0
max-point-size=8
min-point-size=2
model-centroid-x=-1
model-centroid-y=-184
model-centroid-z=65
threshold-primary=48
automatic-threshold=true

[keyboard-shortcuts]
key-alt-center=false
key-alt-toggle=false
key-alt-zero=false
key-ctrl-center=false
key-ctrl-toggle=false
key-ctrl-zero=false
key-index-center=0
key-index-toggle=0
key-index-zero=0
key-shift-center=false
key-shift-toggle=false
key-shift-zero=false
keycode-center=Home
keycode-toggle=
keycode-zero=End

[Accela]
ewma=2
rotation-deadzone=0
rotation-threshold=24
translation-deadzone=0
translation-threshold=49
rotation-nonlinearity=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xf0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0@\0\0\0\0\0\0\0)

[freepie-udp-tracker]
add-pitch-degrees=0
add-roll-degrees=0
add-yaw-degrees=0
axis-index-x=0
axis-index-y=1
axis-index-z=2
port=5555

[modules]
filter-dll=Accela
protocol-dll=freetrack 2.0 Enhanced
tracker-dll=PointTracker 1.1
`

MathijsG, 2016-08-31:
The bug is: It should only stop tracking WHILE the key being HELD.

Now it also triggers stopping of tracking while just pressing the assigned key very shortly. That should not happen at all I guess.

sthalik, 2016-08-31:
Sorry but config won't help. I see in keybinding-worker.cpp:

line 130
```
                for (int i = 0; i < 256; i++)
                {
                    Key k;
                    if (old_keystate[i] != keystate[i])
                    {
                        const bool held = keystate[i] & 0x80;
```

So the keystroke will get delivered.

But then I look in tracker.cpp. Maybe using std::atomic_bool will help.

sthalik, 2016-08-31:
I've added synchronization. It should work. I'll drop a build.

sthalik, 2016-08-31:
We're done in #434

sthalik, 2016-08-31:
We're done here too as this appears to be the same bug. Please comment if it's still broken on build in #434

closed 2016-08-31 by sthalik

tboby, 2020-07-02:
I'm experiencing this bug when I use keybindings with modifiers in them.
If I bind "1", it's fine.
If I bind "Ctrl-alt-1" then "Zero while held" remains held indefinitely. They only ways to resolve it is by stopping and starting, or sometimes by spamming the keybind.

This isn't the issue in #434, more like the opposite.