opentrack/opentrack #423: New feature request

individ2016, 2016-08-22:
I don't know if i'm writing in right place, but. I have an offer for new function. I see that mapping curves work like an dead zones of joystick.. I mean that if i select curve to be slow rotation about start of X and Y axis - they do not work together.. It will be perfect if i could say - be slow only about center (aim-cross), but if i'm looking to the right and move view from up to down i will find that it will be slow about center..
What if i want something about circle gradient mapping.. may be not exactly circle, may be an ellipse?
Well.. its just an offer for future development.

Thank you for a such good thing, that i can use :+1:

DaMichel, 2016-08-23:
I like the idea. 
It would certainly be possible by going from (pitch, yaw) to polar coordinates, applying mapping to the radial component and going back to (pitch, yaw), or something like that ...

sthalik, 2016-08-23:
@DaMichel I don't understand the original poster's issue description. As for nonlinear mapping for the output, sounds good if there's a benefit. Could you explain the poster's description please?

DaMichel, 2016-08-23:
Imagine you set a non-linear mapping for pitch so pitching goes slower when you look at the horizon, so at pitch = 0. Imagine now that you look at something to your left or right. Then, while looking in that direction you pitch up or down, crossing the horizon line. When that happens pitching goes slower due to the mapping, right? But what if you don't want this. What if you only want slower movements when looking forward (pitch and yaw = 0)!
So I'm thinking a mapping as follows would work, given pitch p, and yaw y:
```
# just an example
mapping(r) = r*r if r < 1, else r

r, theta = kartesian2polar(p, y)
r = mapping(r)
p, y = polar2kartesian(r, theta)
```
so in this particular example the movement would appear slow for the circle in the pitch-yaw plane where |y*y + p*p| < 1/4. Beyond that it would be faster, and for  |y*y + p*p| > 1 it would be normal.

individ2016, 2016-08-23:
Well.. My english not so good maybe.. My offer is kind of tagged number 2, instead of 1 (what we have now) on the picture:
[image]

MathijsG, 2016-08-27:
It sounds a bit like this: https://github.com/opentrack/opentrack/issues/292

Without being too specific, is there something possible with defining the curve of an axis based on the value of another? That would create many possibilities for defining behaviour of an axis.

individ2016, 2016-08-31:
I can't imaging how it will be used except only Yaw and Pitch axes.. The sense is just for easy to keep in eye enemy, when you need it, and easy to looking about other space in other time.

sthalik, 2016-09-26:
If we want change about one axis to affect value about other axis, it needs good user interface design. Something not ad-hoc like the #292 freetrack feature request.

We can't expect users to understand polar coordinates either. People were badly taught math and don't event want to look at anything remotely reminding them of school "math" experience.

One thing we could do is allow for one DOF to scale the other.

closed 2018-05-14 by sthalik