opentrack/opentrack #425: UDP sender does'nt work

topiq77, 2016-08-22:
UDP sender does not work with Windows Phone . Last run well in version 2.3.0 RC7.  Tested with Nokia Lumia 820 with a gyroscope on OpenTrack Head Tracking 1.2.0.4

It does not work properly on version 2.3.0 RC10, There is a problem with the axis Yaw.

sthalik, 2016-08-22:
What's with yaw?

topiq77, 2016-08-22:
Raw Tracker DATA YAW Axis value varies with rotation, but Game data YAW Axis value is still 0 build 2.3.0 RC10.
2.3.0 RC11 no response to rotation Raw Tracker DATA YAW Axis value is 90, Game data YAW Axis value is 180. The rest of the position axis indicates 0.
Sorry for my English , I use the translator. 
It's hard to explain the problem in a language other than Polish

sthalik, 2016-08-23:
Ze smartfona to raczej aplikacje obsługują tracker freepie-udp. Upewnij się co dana aplikacja obsługuje.

Dla testów polecam też wyłączyć "center at startup".

sthalik, 2016-08-23:
W rc11 w ogóle nie działa przez głupiego byka. W rc12 zacznie bo już jest poprawka w drzewie.

To jeśli chodzi o sam program. Możliwe, że kompilator wygenerował coś nieprawidłowego. Tyle tylko, że właśnie sprawdzałem prostym skryptem w Pythonie i odbiera poprawne wartości. Możesz użyć tego samego skryptu żeby potwierdzić.

Jeśli któraś z wartości jest stała w danym rc, znalazłeś błąd (rc11 ma zepsuty cały udp tracker). Zmieniają się cały czas bez sensu.

<pre>
import time
import struct
import socket
import random

SOCK_IP = "127.0.0.1"
SOCK_PORT = 4242

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

while True:
    sock.sendto(
            struct.pack('@dddddd',
                -20, 10, 30, # xyz
                random.uniform(-40, 40),  # yaw
                random.uniform(-60, 60),  # pitch
                random.uniform(-10, 10)), # roll
            (SOCK_IP, SOCK_PORT))
    time.sleep(.1);
</pre>