opentrack/opentrack #426: RC11 weird Pitch coupling with yaw/roll - Aruco

503Kento, 2016-08-22:
I'm getting pitch coupling in RC11.  When I first tried it out if I yawed to the right I was suddenly inverted, yawing to the left was fine.  I tinkered a bunch with settings to no avail then tried again with the default.ini.  The yaw coupling was replaced with roll coupling.  rolling my head right gets me inverted; rolling left works as expected.

RC7 and RC10 (with simconnect fix) worked just fine for me.

sthalik, 2016-08-25:
See the warning in aruco article <https://github.com/opentrack/opentrack/wiki/Aruco-tracker>, the last paragraph.

503Kento, 2016-08-25:
@sthalik Thanks, assuming I understand the paragraph correctly I have that in place already as I have the bottom edge of the marker closer to the camera than the top (oblique angle).  I'll play with the angle and see if I can make a change.  As I said, previous releases work just fine with the same hat/marker angle.

sthalik, 2016-08-25:
To have pitch not confuse itself you need as much as 15-20 degrees pitch up or down at all times. That is, +- 15 to 20 degrees is the closest to the center the marker can get.

I'm thinking of a design to alleviate this - rather than putting it as an axis-aligned rectangle, or as top-bottom left-right "diamond-like" corner positions, put it halfway in between. The slant should prevent the breakage and allow for pitch with no restrictions.

503Kento, 2016-08-25:
I have about that amount of angle normally.   Perhaps when I roll that gets fouled up.  

I'll try to rotate the marker a bit and see what happens.

503Kento, 2016-08-26:
Well, I tried all manner of angles for the marker with no success.  I even tried tilting my head forward to make sure I had positive pitch at all times but rotation to the right still flipped the octopus inverted.  All of the same marker tests work as expected in RC7.

sthalik, 2016-08-26:
What's the raw pitch value just before it breaks?

sthalik, 2016-09-16:
@503Kento we had no changes to aruco since rc7. This may have to do with the compiler. See if #454 does anything for you. If not, send a raw camera video that I can replay to see the bug. Just please compress it.

503Kento, 2016-09-18:
Still same Issue with experimental build.

By raw camera video you mean just capture from CL-Eye test?

sthalik, 2016-09-19:
@503Kento yes, but compress it.

503Kento, 2016-09-22:
@sthalik hopefully this is what you need.

https://drive.google.com/open?id=0BxmLU5HrJBobc25QU1BRMjBkS2s

closed 2016-09-22 by 503Kento