opentrack/opentrack #427: PointTracker Threshold Slider Odd Behavior

DaMichel, 2016-08-22:
Try this: 
Start Point Tracker
Open PT Settings dialog.
Move threshold slider back and forth while PT is running.
After a few sec. the tracking will have intermittent failures even for sensible threshold values.
Close Settings dialog. 
Tracking is back to normal. 

This happens with current unstable code under my MSVC build. I don't remember encountering this before. Could it be a unwanted result of your settings refactoring?

sthalik, 2016-08-23:
What kind of intermittent failures?

DaMichel, 2016-08-23:
It just doesn't track any points but just for short moments.
I guess I can investigate it myself a little closer if you cannot readily reproduce it.

sthalik, 2016-08-24:
Try this way: allocate the slider position in `extract_points` into a file each frame. See if values differ.

DaMichel, 2016-08-25:
They do. There are signals from QT which are linked to `setValue` of `base_value`. I'm not touching the slider though.

DaMichel, 2016-08-25:
I think it might be some sort of feedback loop. 
settings::threshold emits value changed signal
slider reacts to value changed and emits value changed itself
settings::threshold reacts to slider value changed and emits value changed
...
I can imagine there being several values "circulating" within QT's messaging queue under the right conditions.

DaMichel, 2016-08-25:
I believe one should only `emit valueChanged` when the value actually changed. Your code is too sophisticated ... I cannot tell if it does so or not.

sthalik, 2016-08-25:
I remember that bug now! It has to do with slot connection type. Try setting both for QSlider to safe_conntype. In tie_setting in value.hpp. If not, see what setting both to direct does.

sthalik, 2016-08-25:
I'll be back home soon but I can't reproduce that race with msvc.

sthalik, 2016-08-25:
If it loops to itself then I have a different fix anyway. Haven't pushed yet.

sthalik, 2016-08-25:
It's only a problem if it loops to old value, i.e. not equal.

DaMichel, 2016-08-26:
Appears to be fixed. Thanks.

closed 2016-08-26 by sthalik