opentrack/opentrack #430: Xbox live camera disconnecting

Jassida, 2016-08-25:
Hi.

Once everything's up and running then there are never any problems but it always takes me a long time to get going for the following reasons:

-starting tracking regularly removes the camera from device manager and I have to unplug and reinsert.
-I must alter the brightness, exposure and contrast every time...it doesn't remember these settings

I am using the latest non stable build and there are no other Xbox live vision drivers I can try.

Is there anything I can do to have it all just work every time I turn my pc on?

Thanks a lot.

sthalik, 2016-08-25:
For the first issue, there's not enough power on the USB controller to service the device. Try moving some devices to the other controller. It's also possibly a driver bug.

For the second issue, it's not opentrack's responsibility; it's the driver's.

Jassida, 2016-08-25:
Thanks.

I think it must be a driver bug as it seems to happen on every USB port, front and back.

closed 2016-08-25 by Jassida