opentrack/opentrack #431: Hatire Arduino translation (X, Y) - very slow.

Ec0nomist, 2016-08-25:
I'm using Arduino inertial tracker. Translations X and Y are linked with head rotation (Yaw and Pitch). In version opentrack -2.3 rc49p7 everything works quickly , the new version rotations (Yaw and Pitch) - fast, translations (X and Y) - slow . Where it is possible to adjust their speed \ smoothing ?
https://www.youtube.com/watch?v=UFKYD1pagcE&feature=youtu.be

sthalik, 2016-09-05:
What happens if you disable Accela?

rizer1980, 2016-09-07:
Same problem.
If i set all bars in accela, nothing changed.
If disable accela - problem is gone

rizer1980, 2016-09-07:
EWMA and Kalman work fine too.

sthalik, 2016-09-16:
@rizer1980 @Ec0nomist see the build in #454. The cap is the same as previously now.

commit 7586930b (2016-09-16): filter/accela: increase max movement speed caps

Closes #452
Closes #431
closed 2016-09-16 by sthalik