opentrack/opentrack #435: OpenTrack not working with PS3 Eye

Petersaber, 2016-08-31:
I built myself a neat IR clip with three IR LEDs, attachable to my headset, with a power supply, neat. I did it because initially I couldn't get faceApi to detect my camera, but PointTracker worked, so I decided to go with that (and OpenTrack).

Now I try to run OpenTrack with PointTracker 1.1 aaaaaand nothing happens. Huh. I switch to FaceTrackNoIR. faceApi 3.2.6. now inexplicably picks up my camera and detects my face... but it's not quite acceptable for gameplay, there are some issues. PointTracker 1.0 (nor 1.1) doesn't even detect my camera now, only shows a black screen, and "currently being used" LED on my PS3Eye doesn't turn on.

When I press Start in OpenTrack, buttons grey out, and "Tracking not started" disappears, nothing appears in it's place. 

I'm lost now.

    Windows 7 64-bit
    CL-Eye drivers 5.3.0.0341
    FaceTrackNoIR 1.7 with PointTracker 1.0 (tried that)
    FaceTrackNoIR 2.0 with PointTracker 1.1 (also tried that)
    OpenTrack 2.3.11 portable (doesn't see my camera at all)
    OpenTrack 2.2 (same as 2.3.11)
    Tried installing ManyCams and using it's virtual camera (doesn't change anything)

    PointTracker keeps telling me CameraInfo 0x0 pixels at 0FPS
    faceApi in FaceTrackNoIR works just fine, and when I use it, red LED "ON" lights up on my camera
    no other setting from faceApi in FTNI works, and the red LED "ON" doesn't light up

I just can't figure out what the hell is wrong, and I really wanted to use my IR clip.

MathijsG, 2016-08-31:
Do you have any webcam image with the CL Eye test program which comes with the CL-Eye drivers?

Petersaber, 2016-08-31:
Yes, CL-Eye Test shows me perfect webcam image at max framerate. Skype also works with that webcam flawlessly, and ManyCam picks it up and also shows me an image. In each case, red LED on my webcam turns on when image starts showing in either of these programs.

sthalik, 2016-08-31:
Hey,

Confirm whether it works with the aruco tracker.

Also see if the development build helps - https://www.dropbox.com/s/2nlk6r5jtkqcrao/opentrack-2.3.0-test-20160831.7z

Petersaber, 2016-08-31:
No, it doesn't work with anything in OpenTrack. 

Using the build provided in the link doesn't change anything. 

Going to aruco tracker settings and clicking "Camera settings" -> Open hangs the whole app. 

[image]

sthalik, 2016-08-31:
Try historical builds before rc7. Whether it works will prove something.

Len62, 2016-08-31:
Since you are using IR LEDs on your clip have you modified the PS3 cam?

Petersaber, 2016-08-31:
@Len62 - not yet, but I can clearly see lit IR LEDs in CL-Eye Test picture. If I have to modify it anyway, I'll do it. Right now I want to be able to use any trackIR software in the first place...

@sthalik - I'll try, I'll post the results

Len62, 2016-08-31:
Are you running OT in admin mode? Also, did you set the camera up in the Point Tracker settings window? One other thing, make sure the CLEye Test software is not running while using OT.

Petersaber, 2016-09-01:
@Len62 I tried running OT as an admin. 

I made sure that only one application accesses the PS3 Eye at a time - and none do, as the "being used" LED isn't turned on. 

PointTracker settings:

[image]

Forsaken-Player, 2016-09-10:
@Petersaber 
I would first uninstall the drivers and make windows delete them. Second, once reinstalled go to CL-Eye Test application and to OPTIONS--> Video Capture Pin and select the desire configuration.  Close that app and verify the red led on the camera has turn off. Then try opentrack. I've been using for very long time this software and never ever failed to me but at first I had to configure that way the video otherwise it would crash.

Good luck from Argentina

@sthalik  Thank you so much for this super app!!!

Petersaber, 2016-09-11:
@Forsaken-Player I've already tried doing that, and unfortunetly it still doesn't work. I can't do any further testing now, I won't have access to my computer for 10 more days (travelling)

sthalik, 2016-09-11:
Try installing K-Lite codec pack Mega and resetting codec priorities in codec tweak tools. But it has low chance of helping.

Once I was faced with that and what I did was reinstall Windows in a clean directory, then install K-Lite codec pack as the first thing.

Petersaber, 2016-09-11:
I'll try that on a different machine.

On my main machine I'm using Shark007 codecs, and I don't really like the idea of installing the mess that K-Lite Mega pack is (or at least used to be). Do you know which codec specifically could be responsible? Maybe I could mess around with Shark007 admin panel.

sthalik, 2016-09-12:
Maybe camera filter.

Forsaken-Player, 2016-09-12:
Did you try every usb port available? With early versions of opentrack, the program on my pc would crash if I'd change ports where the camera was working.

sthalik, 2016-09-16:
See https://github.com/opentrack/opentrack/issues/454 since we had some camera access issues.

Petersaber, 2016-09-20:
@sthalik unfortunetly, no change. PS3Eye still isn't accessed. 

Messing with Shark007 codecs didn't help either. I still can't understand why FaceAPI V3.2.6 can activate and use the camera, but nothing else does.

sthalik, 2016-09-21:
Do you get any input in Skype or ffmpeg?

Petersaber, 2016-09-21:
I'm not sure what you mean. I can use Skype just fine, just had an hour long conversation at 60FPS.

Forsaken-Player, 2016-09-21:
Did you try other user on same pc? Just guessing

Petersaber, 2016-09-21:
I don't have another profile on my PC, and I have no idea how could it affect things.

I've finally grabbed ahold of my laptop. Similar system setup to my PC. Will try to get OpenTrack running there, if it works, maybe it'll help narrow down the problem.

Forsaken-Player, 2016-09-21:
You can always create a user and delete it later. And why? Just a guess, but windows sometimes do things that noone understand. Still it's a guess.

Petersaber, 2016-09-21:
Update: It works on my netbook. Which is bizarre. The same system (same install disc, same updates), the same codec pack (Shark007), and the same software (the same CL driver, the same OpenTrack build)

I'll lose my mind, thanks. 

The image seems to be uspside down, is that normal?

Forsaken-Player, 2016-09-21:
Never happened to me. And I've never installed any codec to make it run. I would try a make a new admin account. I never used installable version of opentrack. All I can say by now.

Petersaber, 2016-09-21:
I had Shark007 codecs on my netbook, v4.0.2 and 5.9.2 on my desktop. Now both have 6.4.9 (clean removal and install). OpenTrack gras PS3Eye on my netbook, but not on desktop. Just additional information. We can safely ignore codecs as a possible cause of the bug.

sthalik, 2016-09-22:
I had my image shifted and distorted when the USB hub was overloaded with high-bandwidth devices, but never put upside down.

With regard to codecs - reinstall Windows cleanly (will move old stuff to `/Windows.old`) and save the time for attempts which wouldn't work anyway.

Petersaber, 2016-09-22:
@sthalik yeah, I'm going to clean my SSD and reinstall Windows 7 from scratch due to unrelated issues. And it's about time I did it anyway. 

I'll report if OpenTrack works afterwards or not.

sthalik, 2016-09-22:
I recommend K-Lite codec pack Mega instead. But this is your choice.

Petersaber, 2016-09-22:
@sthalik I've reinstalled Windows 7 on a clean SSD, installed most of my go-to software, Shark007 codecs, CL Driver for PS3 Eye and OpenTrack works, and grabs PS3Eye when asked to track. 

Currently installing updates, just installed SP1 (my disc is so old it didn't have it).

Unless something I install later on changes anything, we won't find out what was causing the bug.

sthalik, 2016-09-24:
@Petersaber congrats on being able to build your own clips. All this resistor/converter stuff eludes me for now.

sthalik, 2016-09-24:
I wasn't able to fix codec issues on my installs either. The Windows video capture subsystem is a mess. I'm writing new capture code that works with ps3 eye dialog and it's 2k lines. It was 3.5k lines but I removed audio and TV tuner support. However, that audio and TV tuner capture by themselves take 1.5k lines is absurd.

I have ideas how to make the code way simpler and shorter but it's hard to even look at that krappy code. It's like shoveling manure.

closed 2016-09-24 by sthalik

HansMarseille, 2020-05-06:
I had the same problem. Oddly enough, I set my FPS to Default, then put it back to 75 FPS again and it worked fine.

PlayerBush001, 2020-08-19:
I had a similar problem where the camera was only displaying black. Turns out (for some reason) the camera settings had defaulted to 0 exposure. Hope that helps someone else out there