opentrack/opentrack #436: The axes X, Y, Z. Manual tuning curves very rough. Too big step.

olegiy, 2016-09-01:
Opentrack-2.3.0-rc11.  Mapping --> Mapping propeties --> The axes X, Y, Z
Manual tuning curves very rough. Snap -  big step.

sthalik, 2016-09-01:
Mapping window translation smap got changed for rc12, X axis is now snapped to 1. Is Y snap of 5 too big for you?

olegiy, 2016-09-01:
I do not have RC12 ((
To fine-tune want to disable the SNAP.

olegiy, 2016-09-01:
Specifically near zero.

sthalik, 2016-09-01:
Having at snap of at least 1 is good since games break if e.g. pitch is more than 90.

Build: https://www.dropbox.com/s/259fio91i97e0dm/opentrack-2.3.0-test-20160901.7z

Are you from the Russian Sturmovik community? I saw the thread via Google Translate. You can share bugs rather than using old builds. 

cheers

sthalik, 2016-09-01:
@olegiy also check the third way to center, called experimental for cameras.

olegiy, 2016-09-01:
Ок. 
Version opentrack-2.3.0-test-20160901. 
1. All old values "Mapping propeties" returned  to default
2. Manual tuning curves very rough. Big Snap step.
https://yadi.sk/d/QiWZxwF4ufrwj
3. Alignment settings - error not found.

sthalik, 2016-09-01:
I haven't yet written a mapping migration function. it'll come later before rc12. 

The video doesn't open on this wifi but I'll fix it once i get home. 

What isn't found?

olegiy, 2016-09-01:
> @olegiy also check the third way to center, called experimental for cameras.

How does it work?  No anomalies have found.

sthalik, 2016-09-01:
Is it better or the same as the regular method?

It centers each axis separately.

sthalik, 2016-09-03:
Tag relates to original snap issue. Snap now set to 1 for both X and Y.

sthalik, 2016-09-04:
Test build(s) - https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

olegiy, 2016-09-05:
The axes X, Y, Z . Сan you make a snap step even less?

sthalik, 2016-09-05:
They snap by merely one degree now.

olegiy, 2016-09-05:
Only axes X, Y, Z
https://youtu.be/0NaTFYWGCb0

sthalik, 2016-09-05:
Is half of that fine?

With 1 pixel Y on ypr it's too hard to get e.g. exactly 90 pitch.

olegiy, 2016-09-05:
Interested area around zero.

sthalik, 2016-09-05:
@olegiy is this good now? https://github.com/opentrack/opentrack/wiki/experimental-builds

olegiy, 2016-09-05:
Nice!!! Thanks )

sthalik, 2016-09-05:
:)

closed 2016-09-05 by sthalik