opentrack/opentrack #440: Translation compensation toggle button / deadzone / more axis disabling options

Sone989, 2016-09-05:
The problem is, with the option ON, every small pitch/yaw movement (accompanied by added Y/X ) makes it really hard to keep view centered when needed (for example when looking at reticle). Whenever I move my head forward (closer to reticle) and then want to apply a slight pitch/yaw while tracking my target, the translation output disturbs precise control around the reticle.
At the same time, the option is a must in my opinion, so disabling it is out of discussion ;]

I realize this is of course an obvious consequence of applying translation after rotation, BUT:
would it be possible to add more options to translation compensation?
Examples:
- enabling user to select axes taken into account before applying translation (i.e. applying translation after yaw, but not after pitch),
- adding a toggle switch button (i.e. disabling compensation while pressing a button, for when 0 degrees translation is needed - like looking at reticle),
- adding deadzone slider, setting the lowest rotation after which translation compensation comes to effect (probably the worst "solution").

Please forgive me my arrogance, I have no idea about programming and thus know nothing about how difficult these may be. Nevertheless, after months I decided just to ask.

In the end, Opentrack is a wonderfull piece of software and I feel obliged to send You a BIG THANK YOU!

And again, thanks in advance fo any reply!

sthalik, 2016-09-05:
I don't see any arrogance on your part.

The first two options look fine actually. How about just adding "toggle disable compensation" and "toggle disable while held" keystrokes?

However I think your translation mapping is too sensitive. While flying Falcon BMS it didn't interfere with aiming.

Sone989, 2016-09-05:
Looks good to me ;)

Yes, my mapping is quite sensitive, but this is due to conditions I have at my desk. Cannot change too much for now.

sthalik, 2016-09-05:
Is "while held" enough or do you need a toggle also? Which one do you prefer?

Sone989, 2016-09-05:
while held, definitely ;)

sthalik, 2016-09-05:
Or maybe freeze translation while held? Which do you prefer?

Sone989, 2016-09-05:
I think disable (zero) comensation. It would be more intuitive when you need it fast.

sthalik, 2016-09-05:
But if you disable it, it's gonna make a big jump. What if it just froze last compensated value?

sthalik, 2016-09-05:
That is, froze the compensated value and added translation as normal.

Sorry for this long exchange.

Sone989, 2016-09-05:
Yes, I know that. I see it as a "zero" button, which you would use when you need to disable excessive translation fast, so big jump in viewing direction is unfortunately part of it. Freezing on the other hand would require user to precisely center his head first (what may be difficult with high sensitivity), and then press the button (and that means delay).

In other words, disable would be better in my opinion.

Post exchange is not a problem, it's all for good ;)

sthalik, 2016-09-05:
@Sone989 https://www.dropbox.com/s/x20syvaewvuoz3t/opentrack-2.3.0-test-20160905_3.7z

Sone989, 2016-09-05:
Thanks You soo much! ;)

Sone989, 2016-09-05:
Reporting: currently it disables compensation, but does not re-enable it on release of the button (one way toggle, instead of while held).

sthalik, 2016-09-05:
@Sone989 it works for me the same as other "while held" shortcuts. Do other held shortcuts work for you?

Sone989, 2016-09-06:
update: You are right, after testing today I see that the switch is ok, effect while held. It just doesn't work properly with the joystick, meaning Opentrack does not detect release of the joystick button and counts that as still pressed.
I guess that's just the way driver works. Joystick is Thrustmaster T.16000, if that helps.

Sone989, 2016-09-06:
If it's not possible to overrun that, then maybe a normal toggle button instead of while held?

sthalik, 2016-09-06:
It's meant to work with joysticks just fine. Let me investigate. My HOTAS broke and I haven't paid for repairs yet.

sthalik, 2016-09-06:
To make it clear, does it work with other functions?

sthalik, 2016-09-06:
That is, "while held" functions.

Sone989, 2016-09-06:
Just checked, same problem with all "while held" functions.

sthalik, 2016-09-06:
One idea that comes to mind is POV hat breakage. It was implemented relatively recently. What are you binding to? Try binding to pickle for testing.

Sone989, 2016-09-06:
What do u mean by pickle?

sthalik, 2016-09-06:
Missile release. Or bind it to any other button that's not part of a 4- or 8-way hat.

sthalik, 2016-09-06:
That is, only to see if that's what's broken

Sone989, 2016-09-06:
Same goes with all the buttons.

Sone989, 2016-09-06:
For testing I also tried X360 controller buttons, same effect.

sthalik, 2016-09-06:
How do you feel about gitter for opentrack? It's web-based IRC for github projects.

Sone989, 2016-09-06:
That's no problem. I already signed up, sent you a message.

sthalik, 2016-09-07:
It's fixed and confirmed works,

closed 2016-09-07 by sthalik

sthalik, 2016-10-06:
We now have more buttons for tweaking translation enablement.