opentrack/opentrack #444: I don't know why this isnt working

boddahburnout, 2016-09-06:
I have used this software with my older iphone but since deleted and reinstalled since reinstall. Now I cant even get the octopus to do anything at all. I'm justing upd sender/receiver. I may have messed up udp receiver? im using gagagu vr steamer to access the opentrack server.

sthalik, 2016-09-06:
Disable firewall for testing. If it suddenly works, you need to add an exception.

boddahburnout, 2016-09-06:
I all ready have

sthalik, 2016-09-06:
See if there's UDP coming. Ping from both ends, etc.

gagagu, 2016-09-07:
@boddahburnout: Please contact me at gagagu@web.de and tell me exactly what you've done and what settings are you using. I will see what i can do for you.

SnowyWolfy, 2016-09-09:
I'm having this same issue, here are my settings.[http://image.prntscr.com/image/f6edf57e745440cdb54aa50ba35e6614.jpeg](url)
I have also installed the GVRS Sender, using iPhone 6s+ with GVRS App.

sthalik, 2016-09-09:
Protocol-wise, sending over python's `struct` module produces intended results. opentrack can't do anything here.

SnowyWolfy, 2016-09-09:
Although this may be **offtopic** and I have no idea where I can talk about this, I can get the gagagu vr streaming working just fine but the head tracking doesnt. **(this does the same for Trinus, others have working headtracking but it seems to be skippy while the stream is non laggy or vise versa, some stream horribly but headtracking works woudnerfully)**

gagagu, 2016-09-09:
As you see on my release page, my plugin is only compatible with opentrack 2.3.0 release candidate 7 (rc49p7) for Windows. Ich will take a look into the code to create a newer version hopefully today.

sthalik, 2016-09-09:
@gagagu see https://github.com/opentrack/opentrack/issues/425#issuecomment-241696078

The Polish response doesn't matter. Just see the code block.

sthalik, 2016-09-09:
As for rc11 it has a logic error breaking ALL udp.

sthalik, 2016-09-09:
But rc12 is to be released today. If you want, join https://gitter.im/opentrack/Lobby for realtime chat and @-mention me.

SnowyWolfy, 2016-09-11:
Thought id update this, maybe this will help the main thread creator, 
Thanks to @gagagu help, I was able to get the Head tracking + stream via phone working on **2.3.0 Canidate 7** Woot.

Only problem I have, when moving my head in a Pitch form (down - up) sometimes it will be smooth as normal and other times it will stop and skip really far, I tried playing around with some of the filters. 

``To add on top of this, If i adjust the filter Accela in sensitivity anything above 1.5 it will act normal anything lower it will move my view up and down very quickly in a spamming way``

I'm fairly new to VR and how it all works, I just picked it up three days ago, so do not take my problems as a for sure. Thank you all for helping and giving ideas on solutions. Heres my new settings for those who are having issues like mine. **[Here](http://image.prntscr.com/image/615ba4b15e5a48088e83da081ef8b8b1.jpeg)**

sthalik, 2016-09-16:
See https://github.com/opentrack/opentrack/issues/454 for a working recent build.

sthalik, 2016-09-21:
UDP tracker not working in rc11 is fixed in rc12.

closed 2016-09-21 by sthalik