opentrack/opentrack #450: OpenTrack freeze when click start button

FlyMakerBR, 2016-09-09:
OpenTrack rc99p11 is freezing randonly after pres start.

Input: Aruco
Output: freetrack 2.0 Enhanced
Filter Accela

Some days work others dont!

Opentrack is not responding and just that for hours

FlyMakerBR, 2016-09-09:
Freezing now when click input properties too

sthalik, 2016-09-09:
First try https://github.com/opentrack/opentrack/wiki/experimental-builds to see if it's gone.

If it's not gone in the latest build I'm gonna provide some debug instructions.

sthalik, 2016-09-09:
Also state your CPU model (no other hardware details needed).

FlyMakerBR, 2016-09-09:
FX8350 4.0 Ghz

sthalik, 2016-09-09:
Then check the latest build. Backup your config first, after tests replace with the backup. This is because mapping format changed and you'll start with no mappings in the new config. BUT rc12 will migrate your old mappings. BUT if you keep the config from the experimental build it's not gonna migrate.

FlyMakerBR, 2016-09-09:
Thx.

I started with Default preset and worked.

sthalik, 2016-09-09:
@FlyMakerBR can I assume it's fixed, then?

rc12 will come later today.

FlyMakerBR, 2016-09-09:
Freeze again. I hit start -> Stop -> Start -> freeze

sthalik, 2016-09-09:
What's your webcam? Can you replug it? Also wait about 5 seconds since some buggy webcam drivers crash on quick stop/start.

FlyMakerBR, 2016-09-09:
Microsoft HD5000

I will try replug again

FlyMakerBR, 2016-09-09:
The default.ini is missing some info. Show only that:

[modules]
tracker-dll=aruco -- paper marker tracker

[aruco-tracker]
camera-name=Microsoft\xae LifeCam HD-5000

FlyMakerBR, 2016-09-09:
I used Opentrack yesterday without problem. Some random freezes on start but after a minute worked.
Today i have tried for hours

FlyMakerBR, 2016-09-09:
I tried to use virtual cam too.
ManyCam
IpCamera
VirtualCamera

Not effective

FlyMakerBR, 2016-09-09:
All experimental builds not worked.
Deleted default.ini before all tests

FlyMakerBR, 2016-09-09:
I tried all trackers. Only when webcam is needed the program freeze

sthalik, 2016-09-09:
Why would manycam not work?

FlyMakerBR, 2016-09-09:
Just not work.
I select manycam in devices. Hit start and freeze

FlyMakerBR, 2016-09-09:
Tried static image on manycam and always freeze. But show the image on opentrack before freeze.

sthalik, 2016-09-09:
Did you get any windows updates today or yesterday?

Did you install or update codecs?

FlyMakerBR, 2016-09-09:
No changes. Only if have background updates.

But i do not see any update screen even on restart.

sthalik, 2016-09-09:
There's update history in the control panel.

FlyMakerBR, 2016-09-09:
I made a new instalation with setup.exe and now i can start. No freeze

All portable releases not working for me.

I'ts weird. I move to portable versions because problems with setup releases. Now i have to go back to binarized release.

sthalik, 2016-09-09:
It's probably a permission issue. Make sure to assign yourself ownership and full control over all the directories and files inside the portable install directory.

sthalik, 2016-09-09:
What's the problem with the installer? Report it next time rather than working around.

FlyMakerBR, 2016-09-09:
"Make sure to assign yourself ownership and full control"

I made that with all releases.
The problem is not the installer. Opentrack instalation stoped so i move to portable. Now portable stoped i move to new instalation

FlyMakerBR, 2016-09-09:
Now i'm trying to disable outputs but the settings always return to default.
I running with ADM privileges. Something is wrong.

FlyMakerBR, 2016-09-09:
Same with invert option. Always return to disable.

sthalik, 2016-09-09:
These ones are broken in rc11. Fixed in experimental.

FlyMakerBR, 2016-09-11:
Every time i restart computer need to reinstall OpenTrack

sthalik, 2016-09-16:
@FlyMakerBR I had it reproduced. See https://github.com/opentrack/opentrack/issues/454

FlyMakerBR, 2016-09-16:
I found the problem. Is not about reinstall or permissions. Is about multiple monitors

I know this is weird but i made the test right now. If the opentrack shortcut is on my second screen desktop when i hit start the program freezes, everytime in the last 10 tries! But if i put the shotcurt on my first screen the program start normally.

That is the most weird bug i ever seen

sthalik, 2016-09-16:
@FlyMakerBR indeed. But see if the build I linked helps. It has some HiDPI support and actually switches DPI awareness off for the whole program.

Also you can go to qt.conf in the install directory and use values of 0, 1, 2 for the DPI awareness parameter.

FlyMakerBR, 2016-09-16:
Tested on all experimental builds. All freezing on second monitor. DPI Awareness parameter not stopped the problem too

sthalik, 2016-09-16:
@FlyMakerBR what's the Windows version you're using?

sthalik, 2016-09-16:
I think it may have to do whether you allow the taskbar to be shown on the second display or not.

FlyMakerBR, 2016-09-16:
Windows 10 Pro 64 Build 10586

FlyMakerBR, 2016-09-16:
Tried to disable taskbar on second display and freezes still happening

sthalik, 2016-09-21:
@FlyMakerBR thanks for diagnosing the issue. It's serious and I woulda searched the software's components for a longer time to see what the hell's wrong.

I'm closing this issue but we can reopen if you wish to find out why it freezes on another display. We'd have to exchange multiple builds and debug info so it depends on how much time you want to allocate. I'm fine either way.

closed 2016-09-21 by sthalik

sthalik, 2016-10-06:
@FlyMakerBR see #469. We're good now. Get a build at https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa