opentrack/opentrack #451: Question about multiple cameras...

Tux0Topo, 2016-09-09:
So... is there any chance to make it happen?

Forsaken-Player, 2016-09-10:
Would be great!!!

Emton, 2016-09-11:
Do you mean like multiple cameras to increase the FOV?

Tux0Topo, 2016-09-11:
Increase the fov? Didn't think about it, but could be an interesting idea... but no.
I'm talking about using 2 or 3 cams (or more), so the tracker can be identified when we turn more than 90 degrees.

sthalik, 2016-09-16:
Not unless someone sends code for review.

closed 2016-09-16 by sthalik

SkeleCrafter, 2017-03-03:
That's going to require some complex code with all this paralax shiz and comparing values and mapping 3d spaces. I don't even know how to code but it already sounds quite like a huge deal although, PSMoveservice does do this.

sthalik, 2017-03-04:
@SkeleCrafter there are multiple state-of-the-art perspective solvers. We use one already in Aruco.

SkeleCrafter, 2017-03-04:
Aruco is the AR code looking thing right? We could use that for having multiple cameras in that case correct? Open source positional tracking sounds awesome.