opentrack/opentrack #452: Lag/smoothing in Accela

H-Bear-22, 2016-09-12:
G'day, 

I am using 2.3-rc99p11 and seem to have a problem with excessive lag/smoothing.

I am using Joystick Input/Freetrack2.0 Enhanced/Accela and it takes ~2 seconds for an axis to pan from full left to full right. There is no noticeable lag in the joystick properties display.

If I adjust the smoothing slider in the Accela properties it appears to have no effect on the amount of lag/smoothing.

Thank you for your help,

Bear

sthalik, 2016-09-16:
The maximum absolute value in degrees is capped. Can you use Kalman or no filter at all for the joystick device?

sthalik, 2016-09-16:
Maybe we can rescale the input in the joystick tracker. But Accela needs a sensibly capped max value.

sthalik, 2016-09-16:
> If I adjust the smoothing slider in the Accela properties it appears to have no effect on the amount of lag/smoothing.

It's controlled by the sensitivity slider.

sthalik, 2016-09-16:
@H-Bear-22 see #454. The cap is now the same.

commit 7586930b (2016-09-16): filter/accela: increase max movement speed caps

Closes #452
Closes #431
closed 2016-09-16 by sthalik