opentrack/opentrack #454: RC 12 coming soon?

Emton, 2016-09-16:
Hey is RC 12 coming along do you need any testing right now?

sthalik, 2016-09-16:
Actually I do. Thanks for the offer.

Here's the build - https://www.dropbox.com/s/udw5brqordolo6p/opentrack-2.3.0-test-20160916_0.7z

sthalik, 2016-09-16:
Here's a link to all the experimental builds <<https://github.com/opentrack/opentrack/wiki/experimental-builds>>. The dropbox URL might change but the wiki page stays the same.

MathijsG, 2016-09-16:
Already encountered one bug: It starts with no window, only a tray icon.

sthalik, 2016-09-16:
@MathijsG it was meant to do that. Tray is for "game detection" functionality. That is, start opentrack with Windows and forget about it.

We don't limit to single instance. Someone might want to test the UDP tracker/protocol over loopback for instance. In general nothing bad can happen other than mangled data if the same protocol is selected twice.

sthalik, 2016-09-16:
The build is now overwritten (ninja updated) with Accela max speed cap increased to rc7 value.

MathijsG, 2016-09-16:
The previous functionality was with "minimize to tray" was that it was minimized to the tray instead of to the taskbar (like many software that has this functionality), but not even starting up windowed seems quite confusing to me. Very often the opentrack icon isn't even visible in the system tray, because of Windows deciding not to show all items there, so people will open opentrack a few times, having it not show up.

Maybe differentiate "minimize to tray" and "start minimized"?

Emton, 2016-09-16:
Should I be able to use the profiles from previous versions or should I make a new one for testing?

Len62, 2016-09-16:
I'll give this one a test and report back.

sthalik, 2016-09-16:
@Emton profiles should migrate correctly where there are changes, i.e. mappings and selected modules.

sthalik, 2016-09-16:
@MathijsG that makes sense, yes.

sthalik, 2016-09-16:
furthermore, toggling tray off won't raise the main window unless it was hidden to tray.

sthalik, 2016-09-16:
Build overwritten.

I'd like to hear from you if accela with greatly increased max speed cap is worse or what.

Emton, 2016-09-16:
Using Aruco I'm getting this bug where the roll is going crazy. On whatever pitch and yaw input the roll goes crazy and shoots to it's maximum allowed value. If I disable roll then I don't think there's a problem. 
Is that this issue https://github.com/opentrack/opentrack/issues/426 ? Do you need some camera data for this?

sthalik, 2016-09-16:
@Emton send a compressed camera feed video please.

Emton, 2016-09-16:
Here try to see if this is good http://www.filedropper.com/test_161

sthalik, 2016-09-16:
@Emton you're wearing the marker upside down.

Emton, 2016-09-16:
I always had it like that I use the invert options, i have to cause I don't have enough space on the other side of the marker to put the tacks through.

sthalik, 2016-09-16:
@Emton I don't see problems with the aruco tracker. It doesn't "overflow to maximum value". It shows a rotation of +- 180 degrees on the Z axis.

Emton, 2016-09-16:
Hmmm ok well should I try it flipped around? Because I normally use version rc49p7, I invert Yaw, Pitch and Y and haven't seen this problem (that's why I asked earlier if I should be able to keep using older profiles, but I see this even with a new profile) Do you want some video of the Opentrack window too?

Emton, 2016-09-16:
Uh ok I just tried turning the shape 180 and I think it behaves more correct but I'll have to keep testing and redo my settings to figure this out. How does the program decide what the shape is and the orientation? Well sorry if this was incorrect bug I figured since I didn't have this problem in the older version it might be an issue...

FlyMakerBR, 2016-09-16:
Camera properties button missing on the last experimental

sthalik, 2016-09-17:
@Emton aruco shape has rules to decide the marker number and normal orientation.

Glad to know it works.

Len62, 2016-09-17:
Ok, moved my old profiles to a different folder so when I ran this version I could start from scratch. Setup went according to plan and OT running fine. Have a couple of more things to check out and I'll report back. Will move my old profiles back and see how they work in this build. Using a 3 pt IR clip with PS3 camera.

sthalik, 2016-09-17:
Your profiles should migrate. That is, new settings will be created in addition to the old ones, especially for all the mappings.

H-Bear-22, 2016-09-17:
The 'lag' problem I reported is much improved in rc12 with the increase in max speed in accela.

Thank you @sthalik for the fix and also for Opentrack in general!

sthalik, 2016-09-17:
@H-Bear-22 improved or fixed?

MathijsG, 2016-09-17:
Can't we make a statement for PS3 eye users that it doesn't work for them or something? It feels wrong to remove such useful feature (having the camera settings inside the program, so you don't need to install something like skype or something just to change the brightness and stuff) just because a few webcam types won't work. I think many users use other webcams than PS3 Eye.

sthalik, 2016-09-17:
@MathijsG we can actually hide the camera settings dialog whenever ps3 eye is selected.

Len62, 2016-09-17:
Put some of my profiles back and they all worked fine. Had to recalibrate them as my camera position is different. When I restarted using the test profile for this build the mappings went back to default, as did the tracker (aruco). Not sure if this was due to the settings not being saved when I made the profile, or due to it being reloaded. Still have some other things to look at. Thanks @sthalik for all of your work on OT.

sthalik, 2016-09-17:
Paste the config please.

sthalik, 2016-09-17:
Or drop in realtime at https://gitter.im/opentrack/opentrack

Len62, 2016-09-17:
Here's the config file. I have to leave for a bit, will check back later.

[rc23 test.txt](https://github.com/opentrack/opentrack/files/478099/rc23.test.txt)

sthalik, 2016-09-17:
@Len62 looks like the config was empty since it doesn't have any old sections in it.

sthalik, 2016-09-17:
In particular -- older .ini sections aren't removed.

Len62, 2016-09-17:
I re-set the tracker and mappings for the test profile and it loads correctly. Might have just been operator head-space.

sthalik, 2016-09-17:
@Len62 if you want to test-run migrations, delete the `[migrations]` section that has the name of the last migration performed.

Emton, 2016-09-17:
Hey I'm getting some strange Y input. What's happening is I'm pitching up and it increasingly adds Y input. When I pitch down to return to center it throws the Y to a crazy negative value. I compared to old version and don't see this, I also tried some of the experimental options but didn't notice a difference. Does this sound weird? My configuration looks like it should be ok but I never know for sure here's some data.
http://www.filedropper.com/crazyyinput

sthalik, 2016-09-18:
https://github.com/opentrack/opentrack/wiki/experimental-builds

The new build has brought back the "camera options" button as well as remade the Accela configuration. Most of the changes are internal, but also the sliders got clamped to sensible ranges.

There's a migration for Accela and it should work for you.

@Emton it's gonna jump like that cause you're not following the last paragraph of Aruco instructions on the wiki.

sthalik, 2016-09-18:
It's a good idea to publish builds on the issue tracker first, in retrospect. Only after you guys are fine with them I'll release them as tagged.

sthalik, 2016-09-18:
Are you all hung over from saturday night? :) I'll release rc12 on monday evening unless there are further complaints. And be my guest -- nitpick the hell out of it.

Len62, 2016-09-18:
Lol! I just finished putting the 9/18 build through it paces with my set-up. Did not run into any problems, no issues with migrations and accela is working great.

Emton, 2016-09-18:
I can not get any steady view at all using Accela on that 9/18 build (using Aruco and PS3eye) The previous version rc99p11 I was using these settings
[image]
And it works much better. So I don't believe you should limit these values especially when not everyone has to use them.

sthalik, 2016-09-21:
@Emton I'm setting the limits to as much as in the message you deleted.

Smoothing is too invasive at very high values. It definitely can't be 125 ms as in the previous version.

Deadzones are clearly visible to the user when set to a high value. I'm setting them higher now only to avoid regressions for users who depended on them.

We can't do anything with marker pitch, no. I don't even think the new Aruco version would improve. This is geometry. There's no solution for the problem. It's sometimes the solver's fault but the solver is already very good. And in many cases no solver could spot a difference since it's so small.

I had a slanted marker for a longer time, just mount it properly, maybe print a bigger one, and it'll do fine. You'll have 25-30 degrees pitch authority.

sthalik, 2016-09-21:
It's released. Anything I totally forgot?

sthalik, 2016-09-21:
I have a really stupid idea. What if Accela saved the values from the last tenth (or fifth, doesn't matter much) second and compared the variance? If variance is big it means that it's just natural noise. The magnitude of the change matters also of course.

FlyMakerBR, 2016-09-21:
"The camera settings button is back."

Thx for that. I really considered not upgrade without this option.

When opentrack 2.3 come to stable i will post on my simulator pages

Emton, 2016-09-22:
Yea that seems like a good range for those numbers, I've been comparing between 3 and 4 degrees for rotation sensitivity and 4 degrees helps me hold the view more steady at the edges and the middle areas (on the steep part of my curves)

For your idea are you talking about for the "pose approaches identity" or just generally for accela? I think that issue is the biggest problem for Aruco because otherwise it's really good being able to do 6-DOF and you just need a camera and it's seems really responsive. I've been able to minimize it by tweaking some but I can't really add more angle to the shape then what I have at the top of my forehead because the shape will disappear too easy looking up (and looking up is more important than looking down a little) Brim hats can get in the way of the shape and kind of hard to mount the shape on that part of the hat anyway. And same issue too much angle the shape disappears (well moreso if you're using low light)

If your idea would help I think it's worth a try if it doesn't eat performance or affect how the filter behaves at the regular angles. Or some other kind of "logic" that determines when the shape pose is at identity and apply some temporary extra "smoothing" I'm not really programmer-smart enough to say the best way.

Hey I had another idea you might find useful (maybe simple too) if you split the Translation Compensation from X and Y axis, in case you might only use it on the X axis for instance. Ok well thanks for this version it's working really good so far.

sthalik, 2016-09-23:
@Emton your model isn't supported cause you don't follow the last paragraph of the wiki page. The one with bold text.

You can indeed make a marker that works cause my 7 cm marker made out of a cigarette box, duct tape, and cardboard stick worked. I also used it with PS3 Eye which has notoriously bad image quality.

You get these jumps not because of Aruco tracking quality but ignoring model requirements.

sthalik, 2016-09-23:
Alternatively we can

- get rid of jumps
- make Aruco return an absolute pitch value, even if it's a minus negate it

but this is bad.

Emton, 2016-09-23:
Ok if you want to try getting rid of jumps I'll test it out no problem

sthalik, 2016-09-23:
But you should fix your model.

The problem with iterative pose estimation is that it sometimes works both ways, but sometimes it confuses directions. It's better to work predictably on a subset of behaviors then work half the time on all of them.

sthalik, 2016-09-23:
So fix your model, there's no opentrack problem. And you don't need sensitivity of 4 on aruco.

Emton, 2016-09-23:
Ok I should fix the model by tilting it back some more is that what you mean? I have a bit less than 1cm of whitespace around the edges is it better to have more? Uh well I'll experiment with it some more

sthalik, 2016-09-23:
Tilt the upper part backward so if you pitch down you get no less than 22 degrees (for safety margin). You'll get up to 50-55 deg upward pitch.

My marker looked like

<pre>
     [CIG]\\
===========\\
</pre>

where equals sign was decent cardboard. The double `[CIG]` was a pack of smokes to support the marker. The backslashes were the marker. Everything was connected at the right places with duct tape.

If the marker didn't fit I could adjust duct tape on the cig pack to rotate on the X axis (pitch).

Of course it gave in to elements with time. I recommend doing something to the marker paper that will preserve it but not make it totally shiny at the same time. If you find some way please do share.

I thought of making an actual marker of a CD case and white duct tape. But it was tedium and LEDs work. However, this is far more solid setup than the paper thing. A black CD case isn't glossy unlike white duct tape so it even helps with exposure differences. With paper and an ink printer you'll get it all shiny and invisible to tracking with too high exposure.

sthalik, 2016-09-23:
Oh, and I put the `====` between my head and headphones. It made for an awkward space cadet but was very practical.

sthalik, 2016-09-23:
And before you complain about pitch authority, my LEDs overlap since the bottommost one is way too close compared to the other ones. That is, the Z value is too different.

sthalik, 2016-09-24:
Thank you everyone for reporting bugs before rc12 got released. You are credited in the release notes. The build is all good to go and there'll be another thread before `2.3.0` gets released. It's been 4 years without a point release and it's about time.

closed 2016-09-24 by sthalik