opentrack/opentrack #456: Looking Behind in GTA 5

JosephMalmer89, 2016-09-22:
This isn't exactly an issue with the program itself, but I wasn't sure where else to ask this.

I'm using the aruco tracker as the input, mouse emulation as the output and EWMA as the filter. Everything is working as intended, except I ran into a minor inconvenience when playing Grand Theft Auto 5.  When I want to look behind myself in first-person mode, I simply turn my head until the in-game camera is looking behind me, which is what I want. The problem is my head is now stuck in this position if I want to keep it in this view, and if I bring my head back to the center, the camera turns back as well.

I was wondering if there is any workaround for this other than using the mouse or joystick to readjust my camera so I can bring my head back to the center. Would the "translation compensation" solve this? I tried experimenting with it but it didn't seem to change anything.

I apologize if this question isn't appropriate here!

sthalik, 2016-09-22:
In the keybindings list there are bindings to "toggle". Toggling means that data from the tracker is temporarily ignored and opentrack returns the same pose just before toggling.

JosephMalmer89, 2016-09-22:
Unfortunately this doesn't seem to work for me. When I turn my head to look behind myself in-game, I press toggle so the camera stops moving, but when I re-center my head and press toggle again, the camera moves back to where it was before.

I also tried pressing toggle first, then turning my head to look behind in-game, pressing toggle again and re-centering my head, but that only moves the camera very little at a time. I'll probably just have to stick to using the mouse and joystick to adjust my camera so I can re-center my head.

sthalik, 2016-09-22:
Hey,

Toggle freezes the head tracking view at the exact point where you were looking when pressing it. You don't need to recenter after pressing it again, just move your head toward center and it'll resume looking where you're looking then.

JosephMalmer89, 2016-09-22:
Sorry about this, I'm a little confused. It seems whatever I do returns the camera to where it was before.

When you say recenter, did you mean my head, or is there some sort of recenter function? 

Also, what you said seems to work when I am walking/running forward, but if I'm standing still, the camera will swing back when I press toggle again.

sthalik, 2016-09-24:
Go to the options screen. The first tab you see has a list of functions that can be bound to keys.

As for "center", it's meant to center your view while tracking is running. You should bind it to something like "Home" since it's the only important binding.

As for "toggle", there's a version "while held" that stops when you press a button and a simple version that stops when you press the same button the second time.

So take your pose for looking backward, use "toggle". While it's enabled NOTHING will change with regard to the pose occuring when "toggle" was enabled. Look wherever you want and there'll be no change.

When "toggle" stops (look 2 paragraphs above for semantics) the view will suddenly become where you're looking right now.

Is that satisfactory?

JosephMalmer89, 2016-09-24:
I'm sure I've done what you told me correctly, but whenever I stop the toggle after I've moved my head to the center, the camera swings back forward. I want to look behind myself and keep that position so I can center my head again instead of holding it there.

If you own the game, are you able to try this and see if it works for you?

sthalik, 2016-09-24:
But that's how toggle works. You can look backward, press toggle once. When you don't need to look backward anymore, press toggle again and the camera will follow you.

closed 2016-09-25 by JosephMalmer89