opentrack/opentrack #457: A4tech PK-336MB

Alex7272, 2016-09-22:
[cam.zip](https://github.com/opentrack/opentrack/files/486464/cam.zip)
program fall

sthalik, 2016-09-22:
I'm sending you a special build that also pops up a console window. It outputs text describing how it tries to shows the dialog. When it crashes I need a screenshot of the console window. For example it looks like this:

[image]

Please only do it while tracking is stopped because it uses other code path that isn't showing text while the tracker is running!

Here is the build - <<https://www.dropbox.com/s/b7qp23v8qt0jkvm/opentrack-2.3.0-test-20160922-test-dialog.7z>>. Please send me the text.

Alex7272, 2016-09-22:
[image]

sthalik, 2016-09-22:
OK. I'll print more information at the place it crashes. Try this build <<https://www.dropbox.com/s/6k4y9i0i2ih17pa/opentrack-2.3.0-test-20160922-test-dialog-2.7z>>.

Alex7272, 2016-09-22:
[image]

sthalik, 2016-09-22:
Everything looks ok before it crashes. See <<https://www.dropbox.com/s/ocdpu2iaz41ojk3/opentrack-2.3.0-test-20160922-test-dialog-3.7z>>. It does some initialization. It may help but it doesn't most of the time.

Sadly things look fine, we're not making any errors trying to show the window.

sthalik, 2016-09-22:
Also I need to know - does it crash when you start it when tracking enabled? Assuming it still crashes no matter what, can you please try it on any other computer with this camera? Only copy opentrack portable, nothing else is needed.

sthalik, 2016-09-22:
This is my final idea at the moment. Let's try using current Windows language for the dialog and not force the absolute default: <<https://www.dropbox.com/s/lfmlv77pj0qyrb8/opentrack-2.3.0-test-20160922-test-dialog-4.7z>>.

Alex7272, 2016-09-22:
[image] tracking is off. In version 11, I wrote about the error. It contains the working version.

sthalik, 2016-09-22:
Let me make it clear - release candidate 11 is OK but release candidate 12 crashes?

sthalik, 2016-09-22:
Also please tell me if it crashes during tracking. During tracking it shows the dialog using different code.

Alex7272, 2016-09-22:
[image]

Alex7272, 2016-09-22:
It does not work in the 11 and 12 versions of, say now working version. I'm sorry that slowly google translate

sthalik, 2016-09-22:
Is there old version that you used where it works?

Also, when you enable tracking and try show the dialog does it crash also?

Don't worry about translation and thank you for reporting this issue.

Alex7272, 2016-09-22:
[image] start

sthalik, 2016-09-22:
But does the dialog crash when you show it while tracking is started?

Alex7272, 2016-09-22:
[Cam07.zip](https://github.com/opentrack/opentrack/files/486906/Cam07.zip)
working

sthalik, 2016-09-22:
p7 is working. I'll research the differences and try to make a new working version.

Thank you for reporting this problem.

Alex7272, 2016-09-22:
Grateful for the work on this program !!!!!

sthalik, 2016-09-22:
I'm happy you say that.

Please work with me to make it work after p12.

This build tries to do things exactly like p7 <<https://www.dropbox.com/s/6ykv90h34lbogfp/opentrack-2.3.0-test-20160922-test-dialog-5.7z>>.

I could've missed some tiny detail nevertheless.

sthalik, 2016-09-22:
Please check:

- tracking stopped
- dialog

and:

- tracking started
- dialog

sthalik, 2016-09-22:
I no longer check the text info cause it's all OK. But I forgot to hide the console window.

Alex7272, 2016-09-22:
[image] 
[image] also does not work

sthalik, 2016-09-22:
This is rc7 rebuilt right now: <<https://www.dropbox.com/s/zqk9pscpipp82cu/rc7-rebuild.7z>>. Please confirm whether the rebuild works. It's the same code just built with the updated compiler, and right now.

Alex7272, 2016-09-22:
not running, start and stop

sthalik, 2016-09-22:
Try this rebuild. I've also rebuilt the libraries PT depends upon.

https://www.dropbox.com/s/rytvmhn08wi7c7y/rc7-rebuild-v2.7z

Alex7272, 2016-09-22:
unfortunately it does not work, start and stop

sthalik, 2016-09-22:
I'm sorry it took so many builds and it still doesn't work.

Please paste the camera name in the `.ini` file and I'll prevent the dialog being shown on it like PS3 Eye, to prevent opentrack from crashing.

sthalik, 2016-09-22:
This is a hard problem because it's not documented why it doesn't work and it breaks or works at random.

sthalik, 2016-09-22:
Unfortunately even if we used a separate executable for camera work it wouldn't solve problems like this. I tried a DirectShow filter browser program written by someone else and it still had problems accessing some property pages.

Alex7272, 2016-09-22:
So this problem can not be solved?

sthalik, 2016-09-22:
It's a problem for oracle of Delphi. The interface to show the property pages is documented. We're following the documentation. The code that shows the camera page while tracking is started was even written by someone else.

Around p10 we changed the compiler to the one recommended on Windows. Practically the only difference in p7 builds was the compiler. And the old one had serious problems. But both generate Windows executables.

Not only it's very subtle (compiler change) but also an insufficiently documented area of Windows (camera property pages). We don't have source code for Windows and your camera driver.

I have a bad solution - in Windows device manager you can change the driver from list and you should have two supported drivers - one from A4Tech and one generic camera driver. If you change to the generic driver it'll work. But it's a bad solution.

Alex7272, 2016-09-22:
launched Amcap program (comes with native drivers) in the options menu and was able to control the camera settings
[amcap.zip](https://github.com/opentrack/opentrack/files/487084/amcap.zip)

sthalik, 2016-09-22:
"amcap" is a Windows SDK sample. It even works with my PS3 Eye property settings!

Let's do it this way - I'll use amcap code for camera access inside opentrack and it'll work. I've read the sample before and it can also read video data from webcams so it can replace the code we're using entirely.

Thank you for confirming this! I'm renaming the issue and you should have a working build within a few days.

sthalik, 2016-09-22:
For reference here we have the webcam code: https://github.com/pauldotknopf/WindowsSDK7-Samples/tree/master/multimedia/directshow/capture/amcap

Alex7272, 2016-09-22:
unfortunately in this I do not understand (in Russian "dark forest")

sthalik, 2016-09-22:
It's code released by Microsoft for programmers to understand. We can use it in opentrack for camera capture and property pages.

You'll have a build with "amcap" code used for camera within a few days.

Alex7272, 2016-09-22:
this is the best news I wanted to hear !!!

sthalik, 2016-09-23:
Here's my update.

I'm making amcap work by removing code from it.

I removed so far

- saving settings
- recording
- audio

Right now I removee TV tuner support...

This sample is 2094 lines now for `amcap.cpp` and there are more files in different directories.

I don't know why something as simple as video access must be that long on Windows, and that boring to read.

Good news

- you said it works for you
- it even works with PS3 Eye sliders
- it works with our compiler

I'll probably need a helper executable based on amcap to read frames. This isn't that big of a problem.

sthalik, 2016-09-23:
I UNDERSTAND WHY PS3 EYE DOESN'T SHOW THE DIALOG

IT MUST BE PREVIEWING.

If it's not previewing to a real DirectShow window the gain/exposure dialog is all greyed out.

sthalik, 2016-09-23:
Hello @Alex7272,

I have good news.

In the first image I have the video buffer captured into memory without preview, along with PS3 Eye config which never before worked for us!

[image]

and in the second image you have some text printed. Because I lowered the resolution to an extremely small one the buffer size is smaller (the last number). The first number is FPS, but because I lowered it to less than 1 frame per second you can see that it appears very rarely.

[image]

The code is a separate program but I'm doing what can be done to simplify it, and either integrate or make into a helper program. I already found out how to change the resolution and frame rate programmatically.

We'll also PROBABLY have input resolution as well as FPS checked from camera rather than a static list in the PT and aruco trackers. No promises.

Alex7272, 2016-09-23:
Thank you for your work, I will be a tester !!!

sthalik, 2016-09-23:
No problem.

Here's the first build. No video. It'll send the video to opentrack so it's not meant to show any video. I disabled it on purpose.

https://www.dropbox.com/s/rrixdkmzr4ql6by/opentrack-AmCap-20160923_0.7z

The build doesn't work on XP unless you run `EDITBIN.EXE` to change the subsystem. But it'll work because opentrack is built the same way, only that `EDITBIN` is run before publishing.

sthalik, 2016-09-23:
You'll get numbers on the console and it's meant to send them. The first number is a timestamp, the second one is the video frame's size.

Alex7272, 2016-09-23:
[bandicam 2016-09-23 13-29-05-439.zip](https://github.com/opentrack/opentrack/files/489003/bandicam.2016-09-23.13-29-05-439.zip)

sthalik, 2016-09-23:
I'll investigate it. You'll get 4 builds:

- Visual Studio 2010 static CRT
- Visual Studio 2010 dynamic CRT (requires x86 redistributable)
- Visual Studio 15 static CRT (like now)
- Visual Studio 15 dynamic crt (redistributable)

It's possible though unlikely that my removing stuff from AmCap caused this but all I did was remove TV tuner and sound card support.

sthalik, 2016-09-24:
Sorry for the delay. We'll only be doing this after `2.3.0` gold is released.

sthalik, 2016-09-24:
Stupid button

closed 2018-01-04 by sthalik