opentrack/opentrack #458: based on the axes

Borisovich, 2016-09-22:
Hello. Please accept my gratitude for the excellent program.Sorry if my question has already been discussed, it is difficult for me to read English messages.
Now, if you don't include "Translation compensation" the review is excellent. For example:
When the head is at the point of alignment, the rotation axis and Yaw rotation occurs only in Yaw, the rotation in the Pitch axis only by the Pitch axis.
When the head is shifted in the Z-axis forward or backward rotations by the Yaw axis rotation occurs only in the Yaw axis, the rotation in the Pitch axis only by the Pitch axis.
The X-axis and so the same work fine.
Everything is perfect, but there is the inconvenience to look back over the chair. It is necessary to invert the axis X.

This includes the "Translation compensation".
Now looking back well. The offset on the X-axis is working correctly.
But if the head is at the point of alignment, the rotation axis and Yaw rotation occurs only in Yaw, the rotation in the Pitch axis only by the Pitch axis.
When the head is shifted in the Z-axis forward or backward rotations on the Yaw axis,or Pitch, occurs at the same time the displacement and rotation of the axes, X, or U. This is very disturbing.
Is it possible to add "Disable Z axis compensation" additional checkboxes to disable unnecessary dependencies along the axes X, y, Yaw, Pitch?

I tried to construct sentences clear for the translator. If you do not understand, I am ready to explain my question.
Thanks again for the excellent program (I understand that a simple "thank you" is Not enough :))

sthalik, 2016-09-24:
Hey,

Would you rather have yaw, pitch and roll be disabled as option for compensation, or all effect on the X and Y axis?

sthalik, 2016-09-24:
Test build: <<https://www.dropbox.com/s/bjoarnx92xmm7bi/opentrack-2.3.0-test-20160924_0.7z>>.

I haven't yet committed the changes because something may be wrong. Translation compensation is hard to test.

You also get different auto thresholding slider values in PT because in p12 and earlier I confused circle area with circle radius. I'll probably add code to migrate from earlier values but for now let's ensure it works.

sthalik, 2016-09-24:
If your X axis meaning is inverted (though I might've misunderstood) then your camera is set to "mirrored" in its options.

Borisovich, 2016-09-24:
Hi. The checkbox "Disable Y compensation" works perfectly and very useful. But there was the problem that when the view offset along the z axis, the rotation on the Yaw axis is offset from the axis X. But I don't shift the look of the x-axis as the same when look at the Pitch axis, is offset from the z-axis

sthalik, 2016-09-24:
Hey,

If you rotate backward then zoom in you move toward the aircraft seat because you're rotated toward it.

We have no way of remembering which way you want at any given time so rotating changes position in an absolute way.

Are there any other issues regarding translation compensation after the changes I sent you?

sthalik, 2016-09-24:
Let me explain compensation.

Without compensation you translate and rotate. With compensation you rotate and then translate according to where you're looking. This is a very simple option requiring a single line to implement.

Borisovich, 2016-09-24:
I'm still going to use the version with additional checkboxes. This partially addresses my questions. Thank you. I think we have a misunderstanding due to me ignorance of the English language. I will try later to explain through video.

sthalik, 2016-09-24:
Hey,

If the problem can be explained in simple Russian words then please write a message in Russian. A family member knows Russian so I can understand that way.

Borisovich, 2016-09-24:
Отлично. Суть моего вопроса в том, что когда выключен чекбокс "Translation compensation", обзор превосходен.  Когда голова в центре, при вращении по оси Yaw или Pitch происходит только вращение по этим осям. Когда я смещаю взгляд по оси Z и вращаю голову по осям Yaw и Pitch, вращение так-же происходит только по этим осям. Но, к сожалению, очень не хватает компенсации по оси Х.
Когда включен чекбокс "Translation compensation", ось Х работает так, как надо, но теперь если сместить взгляд по оси Z, при вращении по оси Yaw или Pitch происходит так-же смещение по осям Х и Y, хотя я не смещаю взгляд по этим осям. Было бы замечательно, если бы все оси работали так, как будто чекбокс "Translation compensation" выключен, но при смещении по оси Х происходило инвертирование только этой оси, как если бы чекбокс "Translation compensation" включен только для неё.

Borisovich, 2016-09-24:
I reformulate my question: is it possible to add a checkbox "Translation compensation only the offset along the x axis"?

sthalik, 2016-09-24:
Hey,

Despite translating I have problems understanding your intention.

When translation-compensating, rotation and position are used to output new position.

We can either exclude some rotation axis from affecting some position XYZ or only do compensation only for some position XYZ.

If you want to disable changing anything except the X axis, disable Y and Z with checkboxes. It'll be enough.

sthalik, 2016-09-24:
If you want only some rotation yaw, pitch or roll not to affect only some X, Y or Z then specify which and I'll review geometry and usage.

Borisovich, 2016-09-24:
I made a video trying to explain your questions. Turn on the subtitles.
https://youtu.be/3100EJDKSyk

sthalik, 2016-09-25:
@Borisovich what if I added an option to prevent compensation from yaw? That is, none of XYZ would get compensated by yaw. You could still pitch down and move forward, it would move you toward the instruments.

In fact I can add 6 checkboxes total - 3 for disable XYZ, and 3 for disable yaw pitch roll compensation.

sthalik, 2016-09-25:
@Borisovich you should like this one: <<https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa>>

I added 3 more checkboxes. Some of them aren't useful but I added them for completeness, "symmetry" maybe.

sthalik, 2016-09-25:
@Borisovich to make it clear, check "disable yaw" in translation compensation options.

Borisovich, 2016-09-25:
I feel embarrassed that I create so many problems, but additional checkboxes problem of parasitic displacements do not solve. Axles are very interrelated.
If you disable compensation by yaw, then it does not work x compensation and z compensation, are needed. Let me remind you that the parasitic displacement in x and y occur when the head is displaced along the z axis.

sthalik, 2016-09-25:
So you want yaw not to affect X compensation but affect others, and nothing else changed?

Borisovich, 2016-09-25:
> @Borisovich to make it clear, check "disable yaw" in translation compensation options.

If you disable compensation by yaw, then it does not work x compensation and z compensation.

But the compensation in x and y required.

Borisovich, 2016-09-25:
Sorry

Borisovich, 2016-09-25:
I have something that clicked, I did not close the conversation

sthalik, 2016-09-25:
@Borisovich so which rotation axis should disable which position axis?

Borisovich, 2016-09-25:
I would ask people, fluent in English, helped me to explain my questions.
Perhaps the more appropriate question is: Is it possible to create additional checkboxes "Enable compensation for the x-axis only." "Enable compensation only for Z-axis"

sthalik, 2016-09-25:
You can ask for translation if you wish. Or you can draw a sample option window of what should be available to the user.

Borisovich, 2016-09-25:
I'll try to ask so
[image]
[image]
Is it possible to work as the last picture?
[image]

sthalik, 2016-09-25:
@Borisovich,

"Translation compensation" is more properly "rotation mode" in the sense that if it's enabled, view position within the cockpit is relative to rotation.

You can also think of it as "rotation order". With it on, you first rotate and then translate the view. With it off, you first move the view and then rotate it.

Also remember that both rotation and translation are continuous. Something happens when viewing the seat having yaw at +- 180, but it's also geometrically consistent when yaw is +-120, +- 90, +- 30 etc.

What options do we have then? We could choose some translation axis to be excluded from being processed by some rotation axis.

<pre>
  yaw    pitch  roll
X [.]    [.]    [.]
Y [.]    [.]    [.]
Z [.]    [.]    [.]
</pre>

Within these possibilities I think you don't want yaw to affect X then Z isn't neutral. But in order to create a third mode for translation compensation (two being enabled and disabled) we must have a function of yaw and the entire X, Y, Z space. Also what you consider "neutral" Z isn't defined in strict terms. Z value of 0 is surely neutral. But what about 0.5, 1.2, 1.8? Are these neutral too?

If you know the formula that maps X, Y, Z for given yaw value (arbitrary XYZ and yaw from -180 to +180) then let's hear it.

Borisovich, 2016-09-25:
Unfortunately, I do not possess such knowledge. Checkbox "Disable compensation by pitch" please save the program. His work satisfies me. Perhaps it will be useful not only to me. Thank you for your responsiveness.

sthalik, 2016-09-26:
@Borisovich are we OK if we keep the checkboxes like in the version I sent you? They're already in the development version before rc13 that hasn't been released yet.

Borisovich, 2016-09-26:
Pisz po polsku, ja szybciej zrozumiem :)

Borisovich, 2016-09-26:
"Disable compensation by pitch" jest dla mnie wyjątkowo przydatny. Pozostałe dodane w moim przypadku są sporne, ale może komuś się przydadzą

sthalik, 2016-09-26:
Więc niech tak już zostanie. Pozostałe są tylko dla "symetrii" czy spójności. Będzie tak jak masz w buildzie w nastepnym RC i w 2.3.0 final. Niestety to wszystko co możemy zrobić geometrycznie.

Skąd rozumiesz polski? Ja czytam Cyrylicę z liceum bardzo powoli. Miałem czwórkę za czytanie Cyrylicy chociaż słów to tylko "privet rebiata" "da zdrastvuyet tavarishch" i koniec. :)

sthalik, 2016-09-26:
Tak samo asymetryczny mapping dla np. Yaw czy X jest bez sensu a jest dla spójności bo DOF to DOF i jak jest sens dla połowy to każdy DOF ma opcję.

closed 2016-09-26 by sthalik

Borisovich, 2016-09-26:
Białoruś, Grodno. Późno przejrzałem twój profil, komunikacja byłoby łatwiej. Piszę przez tłumacza, ale dobrze rozumiem, co jest napisane. Sukcesów :)

sthalik, 2016-09-26:
Cieszę się, że udało nam się cokolwiek zdziałać z problemem. Pozdrawiam :)

Borisovich, 2016-09-26:
Przepraszam za ciekawość, ale ponieważ teraz może być większe zrozumienie, chciałbym wyrazić jeszcze kilka uwag. Nie znam się na programowaniu, to nie oceniaj surowo, jeżeli dla specjalisty to okaże się dość dziwnie. Wychodzę z założenia, że w czasie, gdy "Translation compensation" jest wyłączona, wszystkie osie pracują samodzielnie, bez wzajemnego wpływu. Jeśli przesunięcia w osi x nie, to i obrót w osi yaw dzieje się bez przemieszczenia, niezależnie od położenia wzdłuż osi z. I przesunięcie w osi x tak samo działa dobrze, niezależnie od położenia wzdłuż osi z. Pozostaje tylko kłopot z kierunkiem przesunięcia w osi x widziana temu. Ale po włączeniu "Translation compensation" włączają się pewne globalne obliczenia pozycji na osiach. W moim rozumieniu, tę funkcję musi zrobić jakiś moduł oprogramowania, lub coś w tym rodzaju, który po prostu zmienił współrzędne na osi x przeciwne, jeśli obrót w yaw osiągnął jakiegoś programowalny próg w stopniach. Prawdopodobnie zmiana położenia wzdłuż osi x przy tym będzie odbywać się skokowo, ale w dziedzinie fizjologii człowiek zwykle najpierw skręca głowę, a potem przesuwa, i ten mały brak będzie prawie niewidoczny. To takie moje nieprofesjonalne refleksje. Jeśli jest to trudne, lub niemożliwe, to dziękuję i za to, co się stało. I za cierpliwość :)

sthalik, 2016-09-26:
Nie, nie ma progu. Obraca a potem zmienia pozycję o pozycję podaną przez tracker ale obróconą przez rotację co tracker podał.

sthalik, 2016-09-26:
Na przykład jeśli obrócisz się 90 stopni w lewo to jak pójdziesz X w prawo to przesuniesz się w kierunku nosa samolotu.

sthalik, 2016-09-26:
W wypadku wyłączonego "translation compensation" jedynie ustawia na pozycję, bez obrócenia jej o rotację.

sthalik, 2016-09-26:
A Ty byś chciał próg w którym się neguje X po przekroczeniu pewnej wartości yaw? Dobrze rozumiem?

sthalik, 2016-09-26:
Ja mam pomysł jak to zrobić ogólnie. Takimi samymi mappingami by się ustawiało wpływ jednej osi na drugą (między 1 a -1, gdzie 0 niweluje działanie całkowicie). Użytkownicy prosili o kombinowanie z zoomem a w ten sposób dałoby się użytkownikowi możliwość wpływania każdą osią na każdą inną.

Borisovich, 2016-09-26:
> A Ty byś chciał próg w którym się neguje X po przekroczeniu pewnej wartości yaw? Dobrze rozumiem?
Tak, coś w tym rodzaju. Pytanie z przyjaznym spojrzeniem za ramię ciągnie się już od dawna, jeszcze w FreeTrack cierpienia. Jeśli uda się znaleźć rozwiązanie, to będzie duży plus dla programu, jak myślę.

sthalik, 2016-09-26:
A dlaczego jest to dla Ciebie problemem w ogóle, z ciekawości? Ja w Falcon BMS rozumiałem, że jak patrzę za siebie i ruszam się w prawo to przesuwam się w lewo kokpitu.

Borisovich, 2016-09-26:
Aktywnie używam osi z, a jeśli jest poruszanie się po niej, mocno wzrasta amplituda drgań widzenia, trudno jest utrzymać wzrok w odpowiednim miejscu. Jeśli zbliżyć się do optycznej, lub oglądaj temu, odejście od oparcia siedzenia, na przykład.

sthalik, 2016-09-26:
W PointTrackerze Z jest numerycznie niestabilne. Zwiększ sensitivity i deadzonę na pozycję w Acceli żeby tego nie było.

Borisovich, 2016-09-26:
Próbowałem to, wzrastał martwe strefy do nieprzyzwoitych wartości. Jeśli jest przesunięcie wzdłuż osi Z, przesunięcia w osiach x i y są bardzo duże, chociaż ja tylko patrzaj głową w yaw i pitch.

sthalik, 2016-09-26:
Czyli jeśli nie byłoby niestabilności Z to byłoby ok? Popatrzę na to ale to nietrywialne polepszyć. Nie rozumiem tych publikacji naukowych.

Borisovich, 2016-09-26:
Nie tylko w osi z do czynienia. Jeśli wykorzystać w osi z, to pojawiają się silne przesunięcia x i y. Na przykład, patrzę do tyłu, w lewo, za siebie, i staram się więcej podnieść spojrzenie na osi y. Jeśli spróbuję przejść wzdłuż osi z. to spojrzenie może drastycznie spaść na osi y. Tak samo jeśli ja pochylił się blisko optycznej, to najmniejsze ruchy głowy powodują silne przesunięcie celownika z centrum. Proszę się nie gniewać się, nie upieram się przy natychmiastowym rozwiązaniu mojego problemu. U każdego zadania istnieje obiektywna cena użyteczności. Jeśli zadanie jest w tej chwili zbyt skomplikowane i bez znaczenia, to jestem szczęśliwy i wdzięczny za to, co się stało.

sthalik, 2016-09-26:
Nie ma powodu bym się gniewał. Rzeczywiście PT ma problem z Z przez wykorzystanie algorytmu "coplanar POSIT" ale ten algorytm ma cechy polepszające działanie z punktami na jednej płaszczyźnie (3w LED-y zawsze są na płaszczyźnie). Zobaczę czy da się coś zrobić małym wysiłkiem.

sthalik, 2016-09-27:
Zastanowiłem się. To nie niestabilność POSIT bo ona jest naprawdę mała tylko kalibracja środka. Jeśli nie masz skalibrowane to zrób ale to pomaga tylko do pewnego momentu bo nie da się wyeliminować całkowicie różnic.

Możesz zrobić jeszcze mniej czułe Z (może być liniowe). To pomoże tak samo.

Borisovich, 2016-09-27:
Zmieniałem ustawienia krzywych i martwych stref, osiągnął dobry wynik. Nie do końca to, co sobie wyobrażałem, ale chyba to moje zbędne fanaberie. Przepraszam i dziękuję.

sthalik, 2016-09-27:
To co najbardziej pomaga na szum to mniej wysokie krzywe. Jesli zamapujesz Z od 0 do 30 mm to będzie o wiele więcej szumu niż od 0 do 60 mm. Na przykład. Zobacz ile gra maksymalnie akceptuje Z i kombinuj do tej wartości.

Jeśli chodzi o deadzone na krzywych to ja zawsze unikam i odradzam.

sthalik, 2016-09-27:
A możemy jeszcze dodać slider "translation nonlinearity" do robienia początku filtra mniej szybkim dla pozycji, tak jak "rotation nonlinearity" robi to w wypadku rotacji.