opentrack/opentrack #459: Phantom 'ALT' key press similar to having windows sticky keys activated.

hairypaulsack, 2016-09-24:
Unfortunately I haven't been able to reproduce this on call, but it seems to happen to me quite often when I fiddle with my hotkey configuration settings. I've noticed that when I go to define keys that ALT will always be included and to get rid of it I have to press ALT (at this point outside of the 'press now to configure hotkey' window as just before this moment was when I attempted to define a hotkey and the phantom 'ALT' hotkey press was included) some number of times or even repeat the process where the phantom key is included and doing it two or three times will eventually unstick the 'held down' ALT key.

Now obviously I don't have sticky keys enabled and this happens when switching keyboards. Not a big deal as it's avoidable with some fidgeting, but was curious if anyone else experienced it or if the dev's wanted me to try and recreate the situation and provide some kind of log or something similar.

This has happened for quite some time through the releases.

sthalik, 2016-09-24:
We don't track the state of the left and right alt keys. Only at a time some non-modifier key is pressed the modifiers get checked and read from there. They're read using the same interface as other keys.

Is your keyboard installing "multimedia hotkey" software as part of the driver? Are you using some "gaming program" sitting in the tray all the time?

sthalik, 2016-09-24:
In case you have "Gigabyte OC Guru" running, uninstall it.

hairypaulsack, 2016-09-24:
No I actually never heard of that and if my assumptions were correct the only similar programs that I've used would be EVGA Precision X and MSI Afterburner for fan curve control and OSD running specs.  I just installed it because I switched my old keyboard out with the new one, but the problem persisted on the old keyboard too. Let me try some random stuff because it sounds like it's only happening to me. Maybe a virtual machine run or doing a selective startup + microsoft components and whatever other dependencies my ps3eye cam and opentrack require. I didn't want to say it earlier because I wasn't sure if my memory was correct (it's been awhile since I used opentrack but I wanted to grab it again and see how the new version was working) and as far as I can tell this happens all the time. Is there some log created or is all that done through debugging?

It's not really a big issue to me I just wanted to throw it your way and see if you guys cared to look into it.

sthalik, 2016-09-24:
Get "sysinternals autoruns" and see startup programs.

sthalik, 2016-09-29:
Do any other programs have similar behavior?

sthalik, 2016-10-20:
@hairypaulsack are you pressing ctrl+alt+del perhaps? See #479.

hairypaulsack, 2016-11-06:
Definitely not on the last one. I'll see if I can reproduce to show you what I'm talking about. It most likely is on my end tho. I tried it out in a guest os and was unable to reproduce the problem using the same hardware

sthalik, 2016-12-24:
Windows doesn't allow for it to get fixed. I get the same with dozens of other games. The XInput library doesn't even deal with keyboard stuff.

closed 2016-12-24 by sthalik

hairypaulsack, 2017-05-18:
I have now verified this problem exists with other users on their hardware. I don't remember if I mentioned it, but usually smashing all the modifier keys (L&R-CTRL, L&R-ALT, etc.) though I'm sure that pressing only the ALT key (to my recollection is the only problematic phantom key) would un-stick / free up / depress the phantom alt key.

sthalik, 2017-05-18:
@hairypaulsack the alt key isn't treated any special. I go over all of the keys and Windows tells me whether it's pressed or not. It tells me it's pressed. There is only special handling of the modifier keys (alt, ctrl, shift, windows) in terms of them being modifiers -- the state is retrieved the same way as for all other keys on the keyboard.

sthalik, 2017-05-18:
@hairypaulsack  I'm not denying it happens for you though. Just-that from opentrack's perspective there's nothing to do.

hairypaulsack, 2017-05-18:
Sorry, so were you saying you know what causes it or just that it happens and nothing to do about it? Curious as to what you were using to see the state of the keys? I'd like to see the states as I play so I can get an idea of the frequency of it happening so that I can make a simple AHK script to press / depress the key at an interval that corresponds to the 'phantom' pressing.

sthalik, 2017-05-19:
@hairypaulsack I'm using a 256-byte array for the keys as Windows wants it. The Alt key is a specific index in the array (between 1 and 256), and the value stays the same.

I think that AHK script is overkill, and especially will break apps with menus. Pressing Alt in notepad changes focus to the menu bar.