opentrack/opentrack #461: translation compensation needs a better name

sthalik, 2016-09-25:
The name for "translate after rotate" is `"translation compensation"`. This name sucks. Please help me and find a better name.

sthalik, 2016-09-25:
@MathijsG @DaMichel @dbaarda  ideas?

MathijsG, 2016-09-25:
Let me think aloud. Something like 'Tracking order' and instead of on off you talk about Pitch,yaw, roll axes first, or X, Y, Z first.

Will keep this in mind when I know something even better :-P

Sone989, 2016-09-26:
Do you want it to be more understandable for an average user (straightforward, even if not exactly precise)?

MathijsG, 2016-09-26:
I never really understand the functionality in-depth. Is it mostly just for looking backward AND zooming? Maybe we then can focus on that functionality in the naming.

Like some kind of option button with a question: "When looking behind, you will zoom-in with fysically moving:

- Away from camera
- Closer to camera

Sone989, 2016-09-26:
Not only.
Notice, that whenever you move your head to the left (raw) in X axis, your avatar will close his head to the left wing. Now let's first rotate backward (180 deg output) - your left wing is now on the right side of your head. This means that X rotation got reversed (simplyfying here) in relation to looking forward.
So, whenever you're looking backwards, moving your head over the left side of the headrest will require you, to translate X to the right. At least that's what your brain will tell you to do. Without compensation your avatar will be "locked" to understanding translation as always looking forward.

With option on, no matter where you look, thetranslation output will get dynamically recalculated to ensure the in-game movement takes into account the direction you are facing. That's how dynamic translation came to my mind. While not that precise, it's for sure a more straightforward name.

I'll keep thinking ;)

MathijsG, 2016-09-26:
So something like "relative left and right" and "absolute left and right"?

Sone989, 2016-09-26:
Yes, exactly. And of course it also corresponds othe translation axes.

Sone989, 2016-09-26:
Actually, relative translation sounds good as well ;)

sthalik, 2016-09-26:
That makes sense, thanks. Relative/absolute translation it is.

closed 2016-09-28 by sthalik

MathijsG, 2016-10-11:
#nitpick but can you also change the name in Options -> Shortcuts the bottom one.

commit 10ae6fb5 (2016-09-28): gui/options: rename "translation compensation" to "relative translation"

Issue: #461
Discussed-with: @MathijsG, @Sone989
Closes #461