opentrack/opentrack #462: Rotation in games now jerky

Len62, 2016-09-25:
I've noticed in the last couple of builds that movement in game is jerky only on the rotations, translation movement is very smooth. Will do some more testing and probably start from scratch. Accela smoothing slider did not seem to help. I'll try the other filters and see what happens.

sthalik, 2016-09-26:
We're now running at constant 250 Hz. Previously there wasn't any accounting for timing errors when sleeping.

sthalik, 2016-09-26:
What's your Accela sensitivity value?

MathijsG, 2016-09-26:
Without changing anything to my profile I find the rotational axes very fluid (tried Rise of Flight) and even while the games doesn't have that much FPS on my system the tracking was very fluid.

sthalik, 2016-09-26:
@MathijsG can you see if that's the case in newest experimental build? Link on wiki.

Len62, 2016-09-26:
Sensitivity is 1.8 for Accela. I'm still working at it and if I can't figure it out I'll post some comparison videos. Happens in both IL2-BOS and DCS World, only games I play. I have not played with the sensitivity but will shortly. I will test using 2.3 rc7 also.

sthalik, 2016-09-26:
Do you use rotation nonlinearity? It's as if it was set to a high value.

sthalik, 2016-09-26:
Also see CPU usage. This also happens when opentrack's not getting enough time slices.

Len62, 2016-09-27:
It was set at 1.5 but when I moved the slider all the way to the left (1) the issue disappeared in DCS World. That was with sensitivity at 1.8. In IL2-BOS it is better but I still need to boost the sensitivity. I think in previous versions the rotation nonlinearity was at the minimum by default and I hadn't noticed it was at 1.1. Will do some more testing w/IL2 and see what I can do.

sthalik, 2016-09-27:
I'll be reducing the slider max value. Can you please check what's the max sensible value there?

I recommend 1.05 -> 1.15.

Len62, 2016-09-27:
Maybe 2 is a good limit. I haven't read anything, or saw any issues raised by any users, so it is hard to know what affect it has for other users.

sthalik, 2016-09-27:
It's too much as shown by your experience. I definitely won't go for more than 1.5. It resembles old Accela versions with gain looking like x^3. It behaves like that with sudden "unstick jump".

Len62, 2016-09-27:
1.5 was my initial thought. Can it be reduced below 1 ? And if so what impact would that have?

sthalik, 2016-09-27:
Below 1 would make initial part more noisy.

sthalik, 2016-09-27:
To elaborate, this is the exponent. `x^1 = x`. Also, `forall x, y in R : x < 1, y < 1, x^y > x`. Having exponent greater than 0 but less than one is exactly taking a root of the value. `1/2` is square root, etc.

sthalik, 2016-09-27:
No problem. We're done for rc13.

closed 2016-09-27 by sthalik