opentrack/opentrack #463: Not receiving any data

DistantStatic, 2016-09-25:
Currently using the HeadTrack android app but OpenTrack simply does not process any of the data nor move the little octopus around. All numbers stay at 0. It is opening the port (through wireshark I can see my computer rejecting the packets to the port when I press stop and accept the packets when I press start) but it seems that OpenTrack is not acting upon the data.

DistantStatic, 2016-09-26:
It seems uninstalling candidate 11 and installing 12 has fixed the issue.

sthalik, 2016-09-26:
True. rc11 has entirelty broken UDP tracker.

lowkeygit, 2017-06-08:
I also have similar problem! I plan to use smartphone to track the head movement. I finished setup and found that All numbers are 0 in the output file.
PC: Opentrack 2.3.1.6;
Smartphone: Samsung S2;
Connect: through Wi-Fi;
I tried many methods but still got no progress. What potential problem might be?
Thanks,

sthalik, 2017-06-08:
Enable both send data and send raw. Then enable debug. You should get the orientation displayed on your phone. Is this the case?

lowkeygit, 2017-06-08:
Yes, it is!

sthalik, 2017-06-08:
Add opentrack to firewall. If doesn't help, disable firewall just for a minute to see if the data follows through. NOTE: don't leave it disabled as a solution!

lowkeygit, 2017-06-09:
Now it works! Thank you.

closed 2017-06-09 by sthalik