opentrack/opentrack #464: oculus rift tracking flips octopus

Jerrow, 2016-09-25:
There is an error with the octopus preview with oculus rift (0.8) tracking. Games with mouse emulation goes also wrong like in the preview. Games with freetrack I have not tested.
see video at 18seconds: https://youtu.be/TGY75gRC6ak

sthalik, 2016-09-26:
Hey,

Please explain what's wrong here. It's not clearly evident from the video.

sthalik, 2016-09-26:
In particular I don't see your raw values in the video.

Jerrow, 2016-09-26:
At 18 seconds I am looking up vertical with the rift, and changing the position of the rift very slowly. It's a bit dark in the video but it is visible. The octopus is for some reason rotating fast at that same time.

sthalik, 2016-09-26:
Check today's build <<https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa>>.

I'm not sure if this is the correct fix hence the build says "dirty".

sthalik, 2016-09-26:
I have another solution but let's try this solution first, it's way more simple if it works.

Jerrow, 2016-09-26:
Still there.

very maybe: I think maybe when looking vertical and then a bit further, something does not get flipped on some sort of axis, or gets flipped what shouldn't. But have zero experience programming 6DOF.

Will try some games later. Hopefully I can get a freetrack game running.

Jerrow, 2016-09-26:
don't know if it is related: but when only looking left/right in the rift, so only the yaw changes something not completely makes sense too.
the image also flips in an instance that can't be seen directly when the color of the octopus changes. It can be noticed if looked carefully/closely at the asymetric molds and tentacles of the octopus(when yaw-ing at the spot where the color changes).

sthalik, 2016-09-26:
See if this helps - https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

It's just a normal case of gimbal lock.

sthalik, 2016-09-26:
I'd like to commit the change and fix other Rift versions but I need confirmation first, thanks.

Jerrow, 2016-09-26:
x,y,z still seem to work but only untill 50 cm away from trackir. roll is too sensitive factor 4. pitch doesn't work (at all i think) .

sthalik, 2016-09-26:
See build `_2.7z`. It should be correct algebraically at least.

sthalik, 2016-09-30:
@mm0zct @Jerrow please test https://www.dropbox.com/s/1xg3oeph07ogpvi/opentrack-2.3.0-test-20160926_2.7z if rotating the Rift over 90 degrees makes gimbal lock to occur. I'd like to release rc13 and it's holding it up.

Jerrow, 2016-09-30:
@sthalik 
https://www.dropbox.com/s/1xg3oeph07ogpvi/opentrack-2.3.0-test-20160926_2.7z
Works, no more gimbal lock

sthalik, 2016-10-01:
@Jerrow thank you. I'm applying it to all currently supported Rift versions and releasing p13.

closed 2016-10-02 by sthalik