opentrack/opentrack #465: UDP sender problem with Yaw axis

topiq77, 2016-09-25:
Oś yaw nie przyjmuje dodatniej wartości podczas obrotu. Startuje od -179 do 0, następnie natychmiast przeskakuje znów do -179. W ostatniej działającej wersji rc49p7 po przejściu przez 0, podczas obrotu wartość osi yaw zwiększa się normalnie w zakresie od -180 do +180, w rc99p12 tylko od -179 do 0 i nagle do -179. Przetłumaczyć tego na angielski nie zdołałem.

sthalik, 2016-09-26:
"raw value" masz takie jak powinny być? Jeśli nie to nie w opentracku problem.

Spróbuj wyłączyć centrowanie przy starcie i powiedz mi co się zmieniło. Muszę ustalić w którym komponencie jest błąd.

sthalik, 2016-09-26:
Napisz mi ten "przeskok" jakie masz "raw value" przed i po błędnym zachowaniu.

sthalik, 2016-09-26:
Nie musi być dokładnie, chcę załapać o co chodzi.

topiq77, 2016-09-26:
Mogę nawet zgrać z pulpitu w formie filmu, tylko jak do domu dojadę. Chyba to będzie najlepsze rozwiązanie.

sthalik, 2016-09-26:
No spoko, zapodaj.

topiq77, 2016-09-26:
https://youtu.be/olLn5wWB_og Do 1:16 działająca perfekcyjnie wersja 2.3.0 RC49P7, potem RC99P12.
Ze względu na wszechobecnego 2 latka, który skutecznie utrudnia jakiekolwiek prace na kompie, nagrane na szybko smartfonem. Najważniejsze czyli parametry są widoczne. YT troszkę uciął górę przy edycji stabilizacji (nagrywane jedną ręką, druga operowała Lumią 820). Na wykresie w RC99P12 kropka zatrzymuje się w pozycji dolnej a potem nagle przeskakuje maksymalnie w górę.

sthalik, 2016-09-26:
W ogóle nie powinno tak być, że tracker wysyła kąt powyżej 180. To jest do poprawienia po odbiorze danych od trackera przed dalszym przetwarzaniem.

sthalik, 2016-09-26:
Powinno być w rc13.

sthalik, 2016-09-27:
Zobacz co się zmieniło <<https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa>>.

topiq77, 2016-09-27:
https://www.dropbox.com/s/iwfwkg8burpy9go/opentrack-2.3-rc99p12-21-g346af3e-dirty%20__%20elite_hd.ini%202016-09-27%2020_06_43.mp4?dl=0

sthalik, 2016-09-27:
Czyli chyba działa, co?

topiq77, 2016-09-27:
Śmiga jak widać na załączonym wideo 👍

sthalik, 2016-09-28:
Dzięki za głoszenie.

closed 2016-09-28 by sthalik