opentrack/opentrack #467: Aruco marker maker not working

NobleWizard, 2016-09-29:
I've tried to use the marker maker but it isn't working and I've tried launching it as an admin and it still doesn't work. It might be working and sending the .bmp file somewhere else but I don't know where.

Thank you for any help given.

Tux0Topo, 2016-10-01:
Hi!!
The program need to have some command lines.
The help describe: "aruco_create_marker <makerid(0:1023)> outfile.jpg sizeInPixels"
Silly question, in the folder there is no some test3.png file? Thats an aruco marker...
Other option is to use this page: http://terpconnect.umd.edu/~jwelsh12/enes100/markergen.html
There, you select the id, the image size, and the frame size.

Hope this be helpfull....

sthalik, 2016-10-01:
Always select a .bmp extension for writing the marker. That utility got built without any image file format support and that's all it can output.

Tux0Topo, 2016-10-01:
The test3.png is inside  contrib/aruco.... That's why i sugested to use the file...

sthalik, 2016-10-01:
That's a fine suggestion. I was referring to the executable.

closed 2018-01-04 by sthalik