opentrack/opentrack #468: Program crashes with A4Tech PK-333MB camera

bubal93, 2016-09-29:
Hello.

I have a A4Tech PK-333MB camera. When I'm trying to open it's settings (via input settings -> camera settings (when avaliable), the Opentrack crashes with general "App stoped working" windows error.

Otherwise it works pefect. 
And if I select my built-in camera (laptop), i've got no crashes.
I've tried like 4 or 5 builds, but the problem remains.

I just want to turn off autoexposure on my cam, because I'm getting 14 FPS max.

System - Windows 10 x64. Runs Opentrack as administrator.

Any workaround for this issue?

sthalik, 2016-09-30:
See if this program works <<https://www.dropbox.com/s/rrixdkmzr4ql6by/AmCap-20160923_0.7z>>. This camera input replacement will get in only after 2.3.0. And yes, the video input library we're using now isn't any good.

We've had reports of A4Tech webcams crashing on the dialog. Send me the camera name ini the .ini file and I'll blacklist it from allowing the dialog.

bubal93, 2016-09-30:
This AmCap 9.0.1 doesnt work. I mean it's working, but I'm getting only white screen for both my cameras.
Although I have AmCap version 9 (from my A4 webcam driver) and it works fine.

This is the camera name from .ini: 
`camera-name=A4` TECH PC Camera V`

So there is no workaround agains this? Any way to set camera exposure off outside Opentrack? 

P.S: I've tested Freetrack and there is no bug with camera settings, but I cannot set global camera options there.

sthalik, 2016-10-01:
No, it's not meant to show the preview. I've already removed the preview functionality on purpose. It's gonna be opentrack's video library rather than show something on the screen.

What you should do - select the camera, then go into "video filter..." and you should see the dialog. You can either see it or it crashes.

You can set it in amcap camera filter. But I need to know whether our filter dialog in amcap crashes.

sthalik, 2016-10-07:
Presently all A4 TECH cameras have the camera dialog button greyed out. This should do until the new implementation.

sthalik, 2016-10-28:
It's gonna be this way at least till #457.

closed 2016-10-28 by sthalik