opentrack/opentrack #469: Opentrack freeze when i try start in game

kzfr, 2016-10-02:
Hi, I have weird problem with Opentrack 2.3rc99p7 to 2.3rc99p13 
I Run the opentrack after reinstalling, click start and work, run game and the game doesnt see opentrack but octopus head work properly.
I close opentrack, (dummy track stay on in task menager) i run game, and open opentrack, click on start and opentrack freeze ( waiting 10min ) and nothing, must close opentrack again.

I try copy the npc.dll to folder with game and sometimes it works and opentrack say what game i run (dcs, warthunder, bos..) Prepar3d work good with fsx simconnect. Sometimes work sometimes not.

sthalik, 2016-10-02:
Copying npclient doesn't do anything since games search for its path in registry.

If the dummy process stays in the background then opentrack didn't stop tracking correctly. But this isn't the cause of the issue.

If the user interface freezes, this isn't at all related to the protocol.

Try unplugging your HOTAS just for testing, then see if it freezes. It may be the actual issue.

sthalik, 2016-10-02:
To be specific, if it doesn't freeze with HOTAS unplugged, we need to fix it so it works. This isn't a workaround.

kzfr, 2016-10-02:
I unplugged every devices on usb (only stay ps3eye and keyboard/mouse) and its same. Freeze without run game, only click on "start".

sthalik, 2016-10-02:
Try selecting a different protocol. Does it still freeze? Do the same for the tracker.

kzfr, 2016-10-02:
This same, but i clear registry and now alone opentrack work properly, But when i start any game before i run opentrack this get freeze. I will record for you how it look, but you need a task manager or something ?

sthalik, 2016-10-02:
Select some other random protocol and see if it crashes.

kzfr, 2016-10-02:
protocol like slot or what?

sthalik, 2016-10-03:
It makes no sense to me. I also don't think there's any problem with the code. It's most likely your system's out of whack. But you found a workaround that works, that's good.

sthalik, 2016-10-03:
protocol is now called output. you have freetrack selected now. select something else to see what's at fault.

kzfr, 2016-10-03:
I delete microsoft drivers and turn off, it look like now http://imgur.com/a/SFEGt but all the time freeze.

sthalik, 2016-10-03:
Zainstaluj Windowsa na czysto. I nie używaj więcej programów typu "registry cleaner" bo nic nie naprawią. Nie dłub w systemie bez zrozumienia.

kzfr, 2016-10-03:
Sam usuwałem ręcznie z regedit'u systemowego wpisy o starych opentrackach, oraz odinstalowałem aktualnego i także ręcznie wyszukałem wpisy w rejestrze wraz z ich usuwaniem.
Wcześniej na wersjach poniżej p6 nie było żadnego problemu. Przy wersji p10 był problem z tym że w ogóle przestał widzieć mi kamerkę ps3eye. Udało mi się ją "wykryć przez opentracka" reinstalując CL-Eye driver'a (oryginalnego!)  Gdy uruchamiałem opentracka p10 byłem zmuszony skopiować npclient'a do folderu z grą aby zaczął działać w grze.

Jak widzisz na załączonych screenach mam za każdym razem 2 pozycje a wcześniej była jedna. Będę dalej próbował coś ogarnąć bo nie mam możliwości przeinstalowania systemu.

sthalik, 2016-10-03:
Nie potrzeba absolutnie kopiować npclienta bo opentrack ustawia w rejestrze podczas użycia protokołu freetrack jego lokalizację. Gry szukają tam gdzie jest w rejestrze podane.

Ściągnij `editbin.exe` jeśli możesz. Odpal z konsoli w katalogu z opentrackiem `editbin /SUBSYSTEM:CONSOLE opentrack.exe`  i odpal opentracka rc6 i rc7 tak zmodyfikowanego. W rc7 zobaczysz coś takiego

- `Tracker:: Applying settings`
-  `Tracker::apply ends`

Pojawiają się oba komunikaty?

Poza tym, zawsze możesz wystartować rc6 bez błędów?

sthalik, 2016-10-03:
Zrób to samo z `editbin` na nowym opentracku i powinieneś pomiędzy tymi dwoma dostać jeszcze

- `pt: camera reset needed`

Czy `apply ends` potem się pokazuje?

kzfr, 2016-10-03:
Powiem Ci lepiej, zanim to napisałeś pobrałem wersje portable, usunąłem wersje instalacyjną z systemu (nie usuwając przy tym nic w rejestrze) uruchamiam i opentrack działa... Its Magic! Spróbuje zresetować komputer i zobaczyć czy znów będzie się zawieszać.

sthalik, 2016-10-03:
@kzfr jeśli ten freeze to coś z opentrackiem to tylko protokół. Ustaw jakiś inny protokół i zobacz czy też przywisa.

kzfr, 2016-10-04:
Nie bardzo rozumie co to jest ten protokół mówisz o wyjściu wejściu w opentracku?

sthalik, 2016-10-04:
Wyjście, tam gdzie masz freetrack dwa zero.

kzfr, 2016-10-04:
Hmm ciężko stwierdzić jednak wcześniej nie wiem czy pisałem nie było tego problemu w p3d poprzez ustawienie flightgear lub micrososft fsx simconnect.

sthalik, 2016-10-04:
Nie chodzi o to żeby działało. Piszesz, że włączysz grę najpierw to zwis. Więc ustaw jakiś bzdurny protokół, włącz najpierw grę, odpal z tym bzdurnym protokołem i zobacz czy też zwiesi.

kzfr, 2016-10-04:
Tak też się zawiesiło jak dałem np. mouse emulation

sthalik, 2016-10-04:
To zrób to z editbin co pisałem. Pokaż konsolę jak się wywali.

kzfr, 2016-10-04:
Tutaj jest dodatkowy problem bo przez wiersz poleceń gdy to wpisuje otwiera się dodatkowe okno które od razu znika.

sthalik, 2016-10-05:
No jak zrobiłeś editbin to tak się dzieje?

kzfr, 2016-10-05:
Otwieram wiersz poleceń w miejscu opentracka gdzie skopiowany jest też editbin, wpisuje komende którą podałeś i otwiera się okno na 1s i po chwili znika. Nie jestem wstanie nic zobaczyć a wiersz poleceń przestaje cokolwiek monitorować.

sthalik, 2016-10-05:
Usuń rozszerzenie `.pdf`. To perfidne rozszerzenie pozwalające dołączyć cokolwiek na githuba.

Ja mam od razu przy "start" te 3 komunikaty wyżej z wątku. Co masz przy zwisie?

[opentrack-2.3.0-test-20161005_0.7z.pdf](https://github.com/opentrack/opentrack/files/511028/opentrack-2.3.0-test-20161005_0.7z.pdf)

kzfr, 2016-10-05:
http://imgur.com/wKVtIUR

sthalik, 2016-10-05:
Skoro to cały interfejs graficzny zwisa to włącz tracker "test" i zobacz czy ośmiornica wije mackami.

sthalik, 2016-10-05:
Dla ścisłości, staram się ustalić podsystem w którym występuje błąd.

kzfr, 2016-10-05:
Dałem Test - waving i ośmiornica lata po ekraniku i nic się nie zawiesza, w momencie kiedy przełączyłem na Pointrack'a wszystko się zawiesiło znów. Mi to wygląda na problem z wykryciem gry. W momencie gdy pierw jest uruchomiony opentrack on bez problemu sobie przypisuje ją do operacji, jednak gdy pierw uruchomię grę już nie może jej odnaleźć przez to się zawiesza.

sthalik, 2016-10-05:
A co jak włączysz tracker Aruco? To też używa kamery. Nie chodzi żeby cokolwiek wykrył na aruco.

A jak włączysz tracker test z grą to co się dzieje? Też zawiesza?

sthalik, 2016-10-05:
Spróbuj zmniejszyć ilość klatek na PS3 Eye do 15 FPS i zobaczymy czy też się zawiesi. Sprawdzam jeszcze jedną hipotezę.

kzfr, 2016-10-05:
15 fps - zawiecha
Aruco nic nie zawiesiło

sthalik, 2016-10-05:
Może to załatwimy przez Skype jeśli masz czas. Taka sama nazwa użytkownika jak tutaj.

sthalik, 2016-10-05:
It looks like we're done. I'll leave the issue open for a while in case it breaks somehow.

closed 2016-11-18 by sthalik