opentrack/opentrack #470: True profiles portability

BlueRaven1975, 2016-10-02:
Hello @sthalik, I could not find any documentation on this one so I think this is not implemented but in my opinion it should be.
As OpenTrack can work as a portable application, it would be nice if it allowed the user to configure where to save the profiles instead of using a subfolder in the Documents directory... or maybe it could save them in the application folder by default and have a checkbox for choosing to save them in Documents instead.
This would make it truly portable between different PCs and would make backups/restores easier, too.

sthalik, 2016-10-02:
As a workaround for now you can use `mklink` to link onto a USB stick or something.

It's not a problem to implement profile as a subdirectory if a file named, say, `is-portable.txt` exists. Switching profile directories at runtime is a can of worms and I won't do it.

Saving the current profile is now always done in the registry. I'd rather not bother with this one. Similarly, game detector uses the registry since it has to switch profiles on its own.

BlueRaven1975, 2016-10-02:
Yes, I already used mklink to move it to my Dropbox folder so not a big issue.
But now I have a doubt... is it already possible to keep the profiles in a subdirectory?
Is it enough to have an is-portable.txt file in the application folder so that the .exe will save and search for its profiles in it?
Can you please explain me better what you mean?

sthalik, 2016-10-02:
No, it's currently not possible. It'd be necessary to do some code changes.

But if you're successfully using Dropbox as your profile directory, what is the problem?

That we have a "portable" archive in addition to an installer, is only that a user can inspect it easily, copy files over from development versions for debugging, etc. opentrack isn't a utility to be kept on a USB stick for diagnostics. Users keep their HOTAS, clip, etc. hardware at home and don't carry opentrack with them.

BlueRaven1975, 2016-10-02:
There's no real problem, just that I'm a bit maniac about not "polluting" my folders with heterogeneous contents and would like to keep all my gaming related profiles and settings in a dedicated folder on a secondary HDD... that's all. :smile: 
And you're right, I would not carry OpenTrack around with me, it's just that I find portable version more safe and comfortable to upgrade as soon as new releases are out: instead of overwriting the previous installation I just rename the current folder, put the new one in place and if all is well get rid of the old one.
In this way I can also revert back instantaneously if something goes weird.

sthalik, 2016-10-02:
If we had more user interface elements for that, it'd add to the learning curve. We already have tons of options and hardly any can be done away with. Adding any more configurable elements needs to be justified in order to get the changes in-tree.

And that you found your way around `mlink` is proof that we don't need to add anything to the user interface. The link is merely filesystem metadata.

closed 2016-10-02 by sthalik