opentrack/opentrack #473: camera center as translation offset

Len62, 2016-10-03:
Haven't seen any comments on this but wanted to let you that I have been using it w/RC's 12 & 13 w/o any issues. I know it's experimental but it works great for me. Was there anything specific you were looking for regarding how it works?

sthalik, 2016-10-03:
I'm unsure as to whether offset should be set to camera origin or perhaps to delta between minimum and maximum yaw. also this option breaks on relative translation.

sthalik, 2016-10-03:
that is, it's relative translation that returns bad values.

Len62, 2016-10-03:
I set it up per the info. Relative Translation disabled. Checked use camera center as translation offset and centering experimental camera device. Tracking is spot on and I have not run into any rotation/translation issues. It is also very easy to adjust the camera offset.

sthalik, 2016-10-03:
my idea is that it shouldn't break relative translation. note to self: check if transpose should only be done in compensation for center as translation offset case.

sthalik, 2016-10-03:
@Len62 if you enable relative translation does it behave abnormally? I'm slightly confused about rotation order in the code.

Len62, 2016-10-03:
Enabling it caused some Z axis movement when I pitched-up. Disabling it for the Z axis works unless I continue to pitch-up past 180, otherwise it worked fine. I think that has to do with my camera location (top of monitor) and clip. I'm building a new clip to match the default dimensions, my current clip has the top and middle LED's a little too close to each other.

sthalik, 2016-10-03:
See if replacing this fine makes it work for the Z value. Remove the `.pdf` extension. I've only added it so that github lets me attach it.

[opentrack-logic.dll.pdf](https://github.com/opentrack/opentrack/files/506371/opentrack-logic.dll.pdf)

sthalik, 2016-10-03:
It kinda makes sense though, pitching up will move less forward but also bit higher.

sthalik, 2016-10-03:
Why do cats want to cuddle up to my face when sleeping on my lap? Kropek even licks my lips.

Len62, 2016-10-03:
The dll made a difference. I only get the Z axis movement if I continue past 180 pitch-up. Makes sense that at that great amount of pitch could affect the Z axis. It doesn't come into play when I pitch down, so I'll see what happens after I get the new clip finished. Cats are such great companions, we don't currently have one but at one time there were four here (My daughters).

sthalik, 2016-10-03:
Great, and thanks for confirmation. Does the new text look remotely sensible to you?

[image]

Len62, 2016-10-04:
I would just add the link to the wiki page back in, otherwise it is fine.

sthalik, 2016-10-04:
@Len62 the wiki page is obsolete now that auto-offset works.

Len62, 2016-10-04:
But it explains how to manually set the offset, unless you are going to remove the manual option.

sthalik, 2016-10-04:
It's pointless to set it manually to the exact center position. Also there are positions where it shows all zeros but doesn't make sense.

sthalik, 2016-10-04:
Actually there's another reason. This instruction was necessary before I fixed the rotation formula. With the current formula there's no need to apply this for normal operation, with regard to yaw/pitch interconnect. My clip is heavily rotated and it doesn't matter.

Len62, 2016-10-04:
Ok, I understand. Anything else I can look at for you?

sthalik, 2016-10-04:
In this case, no. Thanks. If you find any bugs please post them.

Len62, 2016-10-04:
Ok.

closed 2016-10-04 by Len62