opentrack/opentrack #474: Aruco seems broke in RC13

Emton, 2016-10-05:
Pitching up seems to make the octopus go crazy on several axis, but especially pitch (jumping to negative, won't go past a certain amount) Yawing right has similar issue but yawing left doesn't seem to have the problem. RC12 seems to work ok, let me know if you want some data or if someone else has noticed the same thing?

503Kento, 2016-10-05:
I've been having the same issue since RC11 See #426 

I have not had a chance to try RC13 yet.

sthalik, 2016-10-05:
@503Kento your marker's mounted upside down as per roll value. Also recommend you add more pitch since you pitch down up to -8. Don't go below 22 pitch when looking down.

Emton, 2016-10-06:
Well I'm keeping positive pitch, the marker should be mounted the correct direction (last time you told me to flip it which I did) RC12 works ok.

sthalik, 2016-10-06:
@Emton I was referring to his marker. As for yours, send me another video please. Use heavy lossy compression.

Emton, 2016-10-06:
I can send a small video file of raw data that is 7.8mb or I can send some extra videos that show the opentrack window and compares between RC12 and RC13 so you can see what I mean, the total for that is 29mb do you want all the videos or just the small one?

sthalik, 2016-10-07:
Send the raw file please.

Emton, 2016-10-07:
http://www.filedropper.com/rawdata let me know if you don't see the issue I can send these views of the opentrack window that show the problem

sthalik, 2016-10-07:
@Emton read the aruco wiki page and especially the last paragraph. I'll add a link to this wiki page to prevent further confusion like this.

Emton, 2016-10-07:
Buddy I've read the last paragraph of the aruco page over and over I've used open track for months, I used it for months, why does it work in RC12 and not in RC13?

Emton, 2016-10-07:
Do you want to see the video of my open track window? I'm using the EXACT same settings from RC12, in RC13 it simply doesn't work at all...

sthalik, 2016-10-07:
@Emton now I do. Fixing.

sthalik, 2016-10-07:
Unsurprisingly it works now, as per the commit. Get the build. Thank you for finding this before 2.3.0 got released. Now quite a few people use Aruco.

https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

Just confirm it and we're good to go.

sthalik, 2016-10-07:
Good news is that now our opencv changes are easy to rebase. They'll persist with little to no work on my side.

Emton, 2016-10-07:
It works now

closed 2016-10-07 by sthalik

503Kento, 2016-10-07:
Works not for me as well