opentrack/opentrack #475: Android, Oneplus3, apk does not recognize gyro

natrius, 2016-10-07:
Hello, i'm using a oneplus 3 with android 6.0.1and with a slimrom 1.4 installed and the app does not recognize the gyro.
There is definitely a gyro and i checked it too with Sensors Sandbox.

What infos do you need? (i'm sure that was not enough)

sthalik, 2016-10-08:
The Android program enumerates sensors as detected in the system, using the idiomatic way to do so. What else is there?

sthalik, 2016-10-08:
See

- https://developer.android.com/reference/android/hardware/SensorManager.html
- https://developer.android.com/reference/android/hardware/Sensor.html#TYPE_GYROSCOPE

sthalik, 2016-10-08:
Install CPU-Z and look for sensor types. https://play.google.com/store/apps/details?id=com.cpuid.cpu_z&hl=en

natrius, 2016-10-08:
*sigh* without an google account i cannot download this program, whatever i am trying. Even Evozi does not work. 

I downloaded the App Sensors Sandbox via the F-Droid store (Appstore for FOSS Apps) and it lists available sensors on my device.
LSM6DS3 Gyroscope and LSM6DS3 Gyroscope Uncalibrated. If i choose the first one: Version 1, Type 4, Maximum Range 17.453293, minimum Delay 5000 Micro(?)s
And changing livedata.

sthalik, 2016-10-10:
To make it clear -- selecting debug shows you no updates for the gyroscope. Is that correct?

If you ever used the .apk before 6.0.1, did it ever work on the same phone?

natrius, 2016-10-10:
Actually the only thing it shows is Orientation. Everything else remain on zero. I gave the app full permissions for the sensors. And as said, the "Sensors Sandbox" (also available in the play store, should be the same app i'm using) the sensors show up.

Nop, the cellphone is new. I tried it once long ago with a xCover 2 and i think it worked there, but i'm not sure tbh. 

A log via CatLog is not really helpfull i guess?

sthalik, 2016-10-10:
If it shows the orientation, do you get data transmitted? Is it accurate?

sthalik, 2016-10-10:
Select "send raw". I see a logic error. You should get the sensor data with "send raw".

natrius, 2016-10-10:
Oh crap. I thought about that, but never tried it. Shame on me. Is it everytimes needed? Because if - it should be activated by default.
But looks good now.

The other thing, X, Y, Z is not working now, but i think thats due limitations of this technology?

sthalik, 2016-10-10:
Error would accumulate too fast with position tracking. I'll fix the logic error soon.

closed 2018-05-14 by sthalik