opentrack/opentrack #476: Translation Compensation rotation offset

melnik151, 2016-10-08:
Hello! I have a strange problem. When using yaw axis at left or right side of X translation it begins to slide X while my head is still, I am just twisting it. The red cross is on its right place. No X translation is detected. Still it goes.

When I put off "enable" tick at relative translation in output menu it's works fine, but there's inversion of X axis when u look back. So I can play with sliding X when I don't want or with inversion at back. Both variants are not nice. :)

There's a video for that issue. https://www.youtube.com/watch?v=D_SJnMGFbFM

Len62, 2016-10-08:
Are you using the accela filter, and if so what are the settings? Also, how are your curves set-up in the mapping tab?

melnik151, 2016-10-08:
That's accela screenshot, say if it doesn't open. https://pp.vk.me/c604620/v604620360/d8bf/9ClgoTATd8Q.jpg

Yaw 180-20, pitch 180-30, roll disabled, X 70-20, Y 70-10, Z disabled. All just straight diagonal.
Nothing unusual I suppose..

Len62, 2016-10-08:
It opened fine. Try experimenting with setting some dead-space on all the axis in the mapping tab.
[image]

Len62, 2016-10-08:
It looks like you were playing IL2-BOS. You can try my profile for that game and see what happens.
[il2 p13 cam ctr.ini.txt](https://github.com/opentrack/opentrack/files/518002/il2.p13.cam.ctr.ini.txt)
Just delete the .txt from the end of the file name.

melnik151, 2016-10-09:
Dead-space does nothing unfortunately. And your profile seems to work the same(sliding X), but little bit more different way. :(

Len62, 2016-10-09:
What camera are you using?

melnik151, 2016-10-09:
logitech c310

Len62, 2016-10-09:
Do you wear your clip on the right side of your head?

sniperwolfpk5, 2016-10-09:
Is there any issue wearing clip to the right side? just asking as my clip  is setup for the right side.

Len62, 2016-10-09:
No issue with using the right side. I was thinking that maybe melnik151's image was being flipped by the camera, just a thought.

melnik151, 2016-10-09:
Yep, on the right.

Len62, 2016-10-09:
Ok. What happens if you disable the x axis? Just trying to narrow down where the problem may be. Also, your point extraction looks like you are getting between 2 and 3 pixels, can you adjust the point extraction and get 7 or 8 pixels? (not that it may matter but you never know)

melnik151, 2016-10-09:
Disable at all or just in relative mode?

I think my diods are rather small for it.

Len62, 2016-10-09:
Disable it on the output tab on the options menu. Are you using IR leds? You can play with the settings in the Logitech software to try and get the best image that isolates the leds. Maybe zoom the camera in to get closer to your clip. What settings are you using for the camera?

melnik151, 2016-10-09:
So when I disable X it doesn't work. As it should do :D
Yep, IR leds. Settings are standard from skype, I just tuned them a little, not automatic.
https://pp.vk.me/c604620/v604620360/de0d/sah_Z0dHMCY.jpg Sry, it's all in Russian.

I'll find now some logitech software and try with it smth.

melnik151, 2016-10-09:
So nothing changed with zooming and making diodes 5px :(

Len62, 2016-10-09:
Well, I'm stumped at this point. Did you have this problem with earlier versions of OT? What resolution are you running with the camera, and in OT? Try setting both camera and OT resolutions to 640x480.

melnik151, 2016-10-09:
I tried to use OT in July for the first time and don't remember any problems with it.

I have no resolution settings in camera software, only Hrz. I tried both 800/600, 640/480.

Still thx a lot that u participate in solving my issue. May be someone knows what to do.

Len62, 2016-10-09:
It appears you cannot adjust the resolution in the camera, thank Logitech for that. Also, you may want to try and d/l 2.3 rc 10 or 11 and see what happens. I'm sure that @sthalik will respond once he sees this thread.

sthalik, 2016-10-09:
Show the X mapping. Also please record the screen directly. I can't see the raw values.

melnik151, 2016-10-09:
That's it https://www.youtube.com/watch?v=yog7B5wHzvA

sthalik, 2016-10-10:
You can't get rid of Z value increasing while yawing, is that the issue?

- check your model dimensions. they must be accurate.
- camera may have high distortion outside the center. try a different camera, e.g. at friend's place

melnik151, 2016-10-10:
Nope, X value sliding, while I just turning my head and red cross is at the same place.

sthalik, 2016-10-10:
Your raw value isn't sliding. As for relative translation, this is expected behavior. It rotates translation based on rotation. So if you rotate yaw 180, you get -20 instead of 20 X translation etc.

melnik151, 2016-10-10:
Look at this, recorded in-game video. https://youtu.be/dAPya2L6NXU

sthalik, 2016-10-10:
It works like this. If you rotate +-180 then it's gonna negate X. Either disable X for relative translation or relative translation entirely.

melnik151, 2016-10-10:
Hmm.. Thx for answer. I used opentrack this summer with other diodes and don't remember that problem, inversion worked correctly..

sthalik, 2016-10-10:
If you rotate first and then translate then it works intuitively.

melnik151, 2016-10-10:
Okay guys thx for spending your time. Understood everything with this :)

closed 2016-10-10 by sthalik