opentrack/opentrack #477: PT cap video needed

sthalik, 2016-10-10:
I need a cap raw camera feed video. Start centered then yaw as much needed to break WITHOUT dynamic pose.

Give your dimensions together in a .zip file. **Highly compress the video to .wmv or .mp4**.

MathijsG, 2016-10-11:
I think I can help you out soon with this.

MathijsG, 2016-10-16:
With breaking you mean that weird flipping which occurs with using a hat for tracking and you get out of the view port of the webcam?

sthalik, 2016-10-16:
I need a video with center to high yaw. It's possible to remove the need for dynamic pose entirely.

MathijsG, 2016-10-26:
Okay I will try to record this week. I'm playing ETS2 on a daily basis now, having a graphics card again and finishing my DIY gear-shifter.

MathijsG, 2016-11-01:
Is this something you can work with? I don't know how I get just the feed itself while also using it for tracking. https://www.youtube.com/watch?v=3vwv-p3jt0s&feature=youtu.be

closed 2018-05-14 by sthalik