opentrack/opentrack #478: Problem with camera offset.

xAK3Dx, 2016-10-11:
So here's the problem. I used facetrack no IR with point tracker and accela filter about year, and now i want to try out new accela filter, as many ppl sayed it's much better than mk1 version i got in payed ftrack v200... i got cap and ps3 eye camera. In ftrack in pointtracker settings i set camera pitch to 20 degrees and everything was good. In opentrack i can't find solution to camera angle. Angle in otpions menu doesnt give same effect as pitch in ftrack... Example - when i set pitch in Otrack - and move on z axis to tv - i move up, not forward... When i set pitch to -20 , z axis work good , all translations are good, but when i use yaw axis - there is roll coming , just as if camera pitch was on 0... thats pretty much unusable... there is no camera angle in pointtracker setup , and setup from main options doesn't work... what should i try?

sthalik, 2016-10-13:
I get up to 5 degrees roll difference when yawing left to right. Are you sure you have the camera centering mode set in options?

xAK3Dx, 2016-10-13:
i got -40 to +40 roll from left to right ((( Well do you mean ' Centering method'? it is default value , but i tried all 3 of them.. In other options only thing i changed in camera - is pitch angle with no results for roll... It works for transition though - my z axis work good with camera pitch , but it doesn't affect roll on yaw... Or i miss something?

sthalik, 2016-10-13:
What's your neutral rotation? That is, when looking forward at your display.

Compare with an older release, in particular rc7.

xAK3Dx, 2016-10-13:
Neutral roll u mean? it's 0... Where i can find older release? seen here only code, but no installer:

https://github.com/opentrack/opentrack/releases/tag/opentrack-2.3-rc7

sthalik, 2016-10-13:
https://github.com/opentrack/opentrack/releases/tag/opentrack-2.3-rc49p7

xAK3Dx, 2016-10-13:
ok i'll check that evening!

xAK3Dx, 2016-10-14:
Just tested build u gave... same problems occur...

xAK3Dx, 2016-10-14:
okay, tested one more time and got these results: with centering method "relative" i occur no roll with yaw, but now z transition got parasite y transition... see here:
https://www.youtube.com/watch?v=DNnkOHdqf44&feature=youtu.be

sthalik, 2016-10-15:
@xAK3Dx you need to fine-tune the pitch value so that moving forward is treated as moving more-or-less perfectly forward. It's not a bug.

xAK3Dx, 2016-10-15:
I would agree, but 1st - i haven't ghot this issue in freetracknoir, and 2nd - how it could be not error if camera doesn't at 0 angle to clip and moving front to it should give y and z transition without camera transition setup? as i see it - seems that transitions aren't connected with camera offset setting. So when i correct yaw/roll problem - transition problem pops up...

xAK3Dx, 2016-10-15:
At my video at 1:00 u can notice that Y transition doesn't work at all also...

sthalik, 2016-10-16:
The camera angle may not have the same effect in relative mode. All movements are, relative -- to the model's center position and not camera's up, down.

xAK3Dx, 2016-10-16:
So than i should use not relative , but absolute centering method?

sthalik, 2016-10-16:
Yeah.

sthalik, 2016-10-23:
Is it any different if you position the camera on the display?

sthalik, 2016-10-23:
Also does roll change on raw data or only on tracker output?

xAK3Dx, 2016-10-23:
ill try to set camera on tv... but that will be tough... i ll check and reply than.. I use 52" tv as monitor... so canera will be or very low as it is now , or very high , but i 'll try anyway.

sthalik, 2016-10-23:
Please answer the raw data question also.

At the very least it's interesting how moving the camera will affect the data.

closed 2018-05-22 by sthalik