opentrack/opentrack #479: Center TrackIR

jojom4n, 2016-10-13:
On Windows 10, if you press CTRL+ALT+CANC and then you select "Cancel" in the screen that appears, without selecting any other entry, the "Center" shortcut in Opentrack does not work anymore (2.3p11 release).

Len62, 2016-10-13:
Just tested and same thing happens in RC13. Ctl-alt-del then hit cancel and OT center shortcut (shift+joy button on my setup) stops working.

sthalik, 2016-10-14:
I can confirm that. After ctrl+alt+delete the ctrl and alt keys get stuck from opentrack's perspective.

We should be using xinput anyway since dinput is deprecated for a time now.

A workaround is to tap ctrl, alt, shift, having it return back to not held state.

Len62, 2016-10-14:
Ah, did not know about the ctrl-alt-shift! Thanks.

sthalik, 2016-10-15:
I'm leaving the issue open and hijacking it as "use xinput rather than dinput".

sthalik, 2016-11-19:
xinput can't be used. There's no benefit and it doesn't even support keyboards...

sthalik, 2016-11-19:
See: https://en.wikipedia.org/wiki/DirectInput#DirectInput_vs_XInput

closed 2016-11-19 by sthalik