opentrack/opentrack #480: Acella Setting not saving

503Kento, 2016-10-17:
When I make changes to the Acella filter, the slider positions are not saved either in the INI file or when I go back to the dialog.

Here is what I'm doing
1) click the setting button next to Acella filter
2) move sliders to desired position
3) click Ok
4) click setting button again
5) all sliders are back to 0

I can see the file timestamp get updated.  I can see other settings get saved in the ini.  I also can edit the file with notepad without issue.

Originally I had smooth tracking but wanted to experiment with different settings.  Now it behaves as though the filter is not working.  I switched to the Kalman filter which works well.

Len62, 2016-10-18:
I have never seen this issue. Are you running OT in Admin mode? Only thing I can think of, maybe something to do with permissions?

503Kento, 2016-10-18:
Thanks for the reply Len62, I suspected permissions as well but dismissed the idea because the INI file is getting updated and I have the same issue when the app is running (just going in and out of the dialog).  I don't know why I didn't try admin mode but I'll give that a go when I get home.

Len62, 2016-10-18:
The permissions was just a hunch. Since the ini was getting updated otherwise I thought the same as you did.

sthalik, 2016-10-18:
Try changing the sensitivity in the mouse protocol. See if it saves itself.

sthalik, 2016-10-18:
Furthermore, please paste accela's .ini section.

503Kento, 2016-10-19:
running in admin mode does not solve the problem
using mouse emulation also does not solve the problem

Here is the ini snippet before (i.e. clean new ini) attempting to change accela settings and after.

-------------------------Before -------------------
[Accela]
ewma=8
rotation-deadzone=100
rotation-nonlinearity=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xf0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0@\0\0\0\0\0\0\0)
rotation-threshold=99
translation-deadzone=0
translation-threshold=0

[tracker-pt]
active-model-panel=0
automatic-threshold=true
camera-fps=30
camera-name=PS3Eye Camera
max-point-size=20
min-point-size=3
model-centroid-x=134
model-centroid-y=-19
model-centroid-z=26
camera-fov=57

[spline-yaw]
points="@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x5@\x14\0\0\0\0\0\0@\x14\0\0\0\0\0\0@,\0\0\0\0\0\0@4\0\0\0\0\0\0@9\0\0\0\0\0\0@K

\x80\0\0\0\0\0@A\x80\0\0\0\0\0@[\x80\0\0\0\0\0@D\0\0\0\0\0\0@f\x80\0\0\0\0\0)"

[spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x4@\b\0\0\0\0\0\0@\x14\0\0\0\0\0\0@0\0\0\0\0\0\0@I\0\0\0\0\0\0@:\0\0\0\0\0\0@W\xc0\0\0\0\0\0@A

\0\0\0\0\0\0@b \0\0\0\0\0)

[opentrack-mappings]
yaw-alt-axis-sign=false
pitch-alt-axis-sign=false

[alt-spline-yaw]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x4@\"\0\0\0\0\0\0@.\0\0\0\0\0\0@4\0\0\0\0\0\0@K\x80\0\0\0\0\0@7\0\0\0\0\0\0@^

\0\0\0\0\0\0@9\0\0\0\0\0\0@e\xe0\0\0\0\0\0)

[spline-roll]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x4@\x1c\0\0\0\0\0\0\0\0\0\0\0\0\0\0@7\0\0\0\0\0\0@>\0\0\0\0\0\0@C\x80\0\0\0\0\0@Q

\x80\0\0\0\0\0@H\x80\0\0\0\0\0@[\x80\0\0\0\0\0)

[accela-sliders]
ewma=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0@\x12\0\0\0\0\0\0\0\0\0\0\0\0\0\0@>\0\0\0\0\0\0)
rotation-deadzone=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\0\0\0\0\0\0\0)
rotation-nonlinearity=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xf0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0?\xf8\0\0\0\0\0\0)
rotation-sensitivity=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xa9\x99\x99\x99\x99\x99\x9a?\xa9\x99\x99\x99\x99\x99\x9a@\x14\0\0\0\0\0\0)
translation-deadzone=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0)
translation-sensitivity=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xa9\x99\x99\x99\x99\x99\x9a?\xa9\x99\x99\x99\x99\x99\x9a?\xf8\0\0\0\0\0\0)

------------------------------AFTER ----------------------------
[Accela]
ewma=8
rotation-deadzone=100
rotation-nonlinearity=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xf0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0@\0\0\0\0\0\0\0)
rotation-threshold=99
translation-deadzone=0
translation-threshold=0

[tracker-pt]
active-model-panel=0
automatic-threshold=true
camera-fps=30
camera-name=PS3Eye Camera
max-point-size=20
min-point-size=3
model-centroid-x=134
model-centroid-y=-19
model-centroid-z=26
camera-fov=57

[spline-yaw]
points="@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x5@\x14\0\0\0\0\0\0@\x14\0\0\0\0\0\0@,\0\0\0\0\0\0@4\0\0\0\0\0\0@9\0\0\0\0\0\0@K

\x80\0\0\0\0\0@A\x80\0\0\0\0\0@[\x80\0\0\0\0\0@D\0\0\0\0\0\0@f\x80\0\0\0\0\0)"

[spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x4@\b\0\0\0\0\0\0@\x14\0\0\0\0\0\0@0\0\0\0\0\0\0@I\0\0\0\0\0\0@:\0\0\0\0\0\0@W\xc0\0\0\0\0\0@A

\0\0\0\0\0\0@b \0\0\0\0\0)

[opentrack-mappings]
yaw-alt-axis-sign=false
pitch-alt-axis-sign=false

[alt-spline-yaw]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x4@\"\0\0\0\0\0\0@.\0\0\0\0\0\0@4\0\0\0\0\0\0@K\x80\0\0\0\0\0@7\0\0\0\0\0\0@^

\0\0\0\0\0\0@9\0\0\0\0\0\0@e\xe0\0\0\0\0\0)

[spline-roll]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x4@\x1c\0\0\0\0\0\0\0\0\0\0\0\0\0\0@7\0\0\0\0\0\0@>\0\0\0\0\0\0@C\x80\0\0\0\0\0@Q

\x80\0\0\0\0\0@H\x80\0\0\0\0\0@[\x80\0\0\0\0\0)

[accela-sliders]
ewma=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0@)\0\0\0\0\0\0\0\0\0\0\0\0\0\0@>\0\0\0\0\0\0)
rotation-deadzone=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xeb\x33\x33\x33\x33\x33\x33\0\0\0\0\0\0\0\0@\0\0\0\0\0\0\0)
rotation-nonlinearity=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xf0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0?\xf8\0\0\0\0\0\0)
rotation-sensitivity=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xf1\xc2\x8f\\(\xf5\xc3?\xa9\x99\x99\x99\x99\x99\x9a@\x14\0\0\0\0\0\0)
translation-deadzone=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0)
translation-sensitivity=@Variant(\0\0\0\x7f\0\0\0\x16options::slider_value\0?\xa9\x99\x99\x99\x99\x99\x9a?\xa9\x99\x99\x99\x99\x99\x9a?\xf8\0\0\0\0\0\0)

503Kento, 2016-10-19:
Also, FWIW, I'm using the build referenced in issue #474

sthalik, 2016-10-19:
It's not as much whether mouse emulation works. Use the mouse emulation slider and see if its changes are broken the same as Accela's.

sthalik, 2016-10-19:
Also back up the config and see if removing all lines containing `options::slider_value` change anything.

503Kento, 2016-10-19:
Ok, sorry I misunderstood your direction.  I'll try changing the mouse emulation slider and deleting the "options::slider_value" lines and let you know.

sthalik, 2016-10-19:
See if this works - https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

I can't reproduce it despite not having made any changes.

503Kento, 2016-10-21:
I installed this version on my work PC as well as RC13 both work as expected.  So, it's definitely something specific to my Sim PC at home.  Hopefully I'll get some time to figure it out this weekend.

503Kento, 2016-10-21:
Intel i7-4790K; 8GB RAM, Win7 Pro running from an SSD

503Kento, 2016-10-22:
I did a complete uninstall including the opentrack-2.3 directory holding ini files and reinstalled RC13 as administrator.  I was able to copy my previous INI file and it accepts updates to the Acella filter.

Everything works as expected now.

Thanks for the help.

closed 2016-10-22 by 503Kento