opentrack/opentrack #481: Aurco solver goes crazy with left yaw and pitch up (after translation)

Emton, 2016-10-19:
Hey the octopus goes nuts when yawing left (past translation) then pitching up. Yawing right (past translation) and pitching up works GOOD (just the left has trouble). 
Here's a video of the opentrack window, PLEASE don't tell me to read the wiki my shape stays at positive pitch except when the solver is going crazy (I center the view at +20 raw pitch I can go NO higher without losing all the pitch space I need to look up (the solver loses the shape around +70 raw pitch))
Let me know if you want raw data
http://www.filedropper.com/solvergoescrazywithleftyawandpluspitch

sthalik, 2016-10-19:
Is it different than #474?

Emton, 2016-10-19:
It is similar to #474 only now instead of the solver getting confused pitching up while I'm looking forward, now it's confused pitching up while translated (looking back) (but only with yawing left, if I yaw right, translation, pitch up, it works ok. So the problem is sort of similar except the axis have just changed it seems.

Emton, 2016-10-20:
Also I compared to rc49p7 I can post some video of it working correctly there if you want.

scarystuff, 2016-10-23:
rc21p11 is the last version the Aruco tracker worked correctly in. Not sure what happened after that.

sthalik, 2016-10-23:
Give me a raw video so I can reproduce it.

Emton, 2016-10-23:
Do you want some raw video of this issue now or did you get it already?

Emton, 2016-10-23:
Here's some raw video http://www.filedropper.com/solverconfusedpitchinguptranslated

sthalik, 2016-11-15:
This is the first test build but it's not fully working --https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

Some rotation and translation axis are swapped. I need to know which. I also need to know how calibration performs.

Emton, 2016-11-15:
I'm not sure, looking forward there is constant full Roll 180 in the raw data, it jumps to negative a lot. Can you tell me what to look for. I noticed the camera Hz is stuck fluctuating between 30 and 40 Hz, instead of the regular 75 for PS3eye cam. How should the calibration work I see the purple dot can move around but I don't know exactly where it should be positioned.

sthalik, 2016-11-15:
And it wasn't upside down for you in a working build?

Please send me a working video with all XYZ, ypr tested. Thanks.

Emton, 2016-11-15:
No it's not upside down in the fallback version I use (rc49p7) I couldn't really get any of the axis tested the octopus is just all over the place. I'll try to make a video later today if you need it.

sthalik, 2016-11-15:
I need a raw camera video to get the signedness correct. Got half a dozen videos where some were filmed of a marker upside down, too close to neutral pose, etc...

Emton, 2016-11-16:
Here is this correct? http://www.filedropper.com/aruco  Also I only tried it quick but when I flipped the marker upside down (different from this video) the octopus behave more correctly. This video is how I use it in rc49p7 and it works fine.

sthalik, 2016-11-16:
Okay.  Also roll is not 180 in p7, is that correct?

Emton, 2016-11-16:
Right, roll is normal in rc49p7. In that 11-15 version just about everything is messed up, but when I flipped the marker (the WRONG way) and inverted the pitch, it worked closer to normal (still had the original bug except it flipped to the right side.

But I want to point out the solver has no trouble operating at negative raw pitch. It only gets confused at identity pose, I made a suggestion for the option: As the solver approaches identity pose ZERO the output (game data) like on a curve, Then wear AND center the marker at identity pose. You would have complete freedom of movement. I understand it's an aruco limitation but sometimes you can work around it.

sthalik, 2016-11-16:
What original bug again?

sthalik, 2016-11-16:
If the solver is still broken then the changes need to become undone.

sthalik, 2016-11-16:
It'll probably work with pose close to identity if marker is normally far from center of the camera's image.

Emton, 2016-11-16:
The original bug is "Aurco solver gets confused with left yaw and pitch up", that was for version rc99p14 now the test version 11-15 is basically back to the problem #474. But if I flip the marker upside down it sort of works, but the bug becomes "Aruco solver gets confused with right yaw and pitch up"

I have to go back and check rc99p12 and compare that to rc49p7

Emton, 2016-11-16:
I tested all the versions back to rc49p10 and they are all bugged similarly to #481. In #474 I had said RC12 worked correctly but I hadn't used it long and didn't see the bug yet, also I skipped a few of those versions before those.

sthalik, 2016-11-17:
It may be something else. It's hard for me to believe opencv update causes that.

I'm disabling DLS pose iteration for testing

https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

Emton, 2016-11-17:
I think that gets rid of #481 bug but the raw data rotation still has 180 and the camera Hz fluctuates between 30 and 40 or so Hz, When the Hz is low the exposure gets brighter and the black area of the shape gets too bright, marker gets lost, so it's making it hard to see what's going on, it should be 75 Hz

sthalik, 2016-11-17:
I replaced it with the old aruco version but with crash fixes and other minor stuff applied. All experimental changes since rc7 went away.

See the last URL.

Just note that you shouldn't use over 60 Hz (or over 120 for 320x240) with Aruco due to the visible distortion.

Emton, 2016-11-17:
Ok this version seems to work pretty good. Did this one get the optimization with occluding the video when opentrack is minimized?

sthalik, 2016-11-18:
No. The optimization breaks the entire program with multiple displays.

sthalik, 2016-11-18:
So are we good for release?

Emton, 2016-11-18:
Yea if that's what you want. I was wondering about the calibration for Aruco should I be using it, I see I can move this purple dot around, do you know if there's a good place to position it or just keep it in the middle of the marker?

sthalik, 2016-11-18:
For calibration, yaw and pitch, then pitch on its own at various yaws.

The important thing here is to do it quickly without pauses so calibration doesn't get biased by certain poses being more common.

closed 2016-11-18 by sthalik

sthalik, 2016-11-18:
@Emton thanks for testing the builds.

scarystuff, 2016-11-18:
@Emton regarding calibration. I calibrate so the dot does not move when I look around. My thinking is that the dot represents the axis point so it should be still when you do yaw and pitch. If the dot is in the middle of the tracker, then the tracker would be the axis point and that would be weird.
Glad you guys fixed the issue, I will test it out later myself.

commit 6844ab95 (2016-11-18): tracker/aruco: back out changes since rc7

However, include some crash fixes and minor changes.

Fixes #481

Reported-by: @Emton
Testing-by: @Emton