opentrack/opentrack #482: Hot key/Center and Zero While held

BECTHuK, 2016-10-22:
Hi! Thanks for a great project! How to make releasing hot button "Zero While held" was the automatic alignment of the sight in the center, without additional pressing hot key "Center"?

sthalik, 2016-10-22:
That's what the `toggle` button does.

BECTHuK, 2016-10-22:
it is better when pressing "hot buttons" is a centering and retaining sight in the center, and when you release the controls and the countdown starts with the same center, regardless of the previous shift head position.

sthalik, 2016-10-22:
I don't understand this, please explain.

BECTHuK, 2016-10-22:
I do not speak English I write through the translator, sorry. I came out writing on 2 teams. When the clamping function is activated "Zero While held", when released triggered "Center". These functions are located in the tab "shortcuts".              
                
            
                
            
                
            
                
            
                
            
        https://translate.yandex.ru/?text=%D0%AF%20%D0%BD%D0%B5%20%D0%B3%D0%BE%D0%B2%D0%BE%D1%80%D1%8E%20%D0%BF%D0%BE%20%D0%B0%D0%BD%D0%B3%D0%BB%D0%B8%D0%B9%D1%81%D0%BA%D0%B8%20%D0%BF%D0%B8%D1%88%D1%83%20%D1%87%D0%B5%D1%80%D0%B5%D0%B7%20%D0%BF%D0%B5%D1%80%D0%B5%D0%B2%D0%BE%D0%B4%D1%87%D0%B8%D0%BA%2C%20%D0%B8%D0%B7%D0%B2%D0%B8%D0%BD%D0%B8%D1%82%D0%B5.%20%D0%AF%20%D0%B2%D1%8B%D1%88%D0%B5%D0%BB%20%D0%B8%D0%B7%20%D0%BF%D0%BE%D0%BB%D0%BE%D0%B6%D0%B5%D0%BD%D0%B8%D1%8F%20%D0%B7%D0%B0%D0%BF%D0%B8%D1%81%D0%B0%D0%B2%20%D0%BD%D0%B0%20%D0%BE%D0%B4%D0%BD%D1%83%20%D0%BA%D0%BD%D0%BE%D0%BF%D0%BA%D1%83%202%20%D0%BA%D0%BE%D0%BC%D0%B0%D0%BD%D0%B4%D1%8B.%20%D0%9F%D1%80%D0%B8%20%D0%B5%D1%91%20%D0%B7%D0%B0%D0%B6%D0%B0%D1%82%D0%B8%D0%B8%20%D1%81%D1%80%D0%B0%D0%B1%D0%B0%D1%82%D1%8B%D0%B2%D0%B0%D0%B5%D1%82%20%D1%84%D1%83%D0%BD%D0%BA%D1%86%D0%B8%D1%8F%20%20%22Zero%20While%20held%22%2C%20%D0%BF%D1%80%D0%B8%20%D0%BE%D1%82%D0%BF%D1%83%D1%81%D0%BA%D0%B0%D0%BD%D0%B8%D0%B8%20%D1%81%D1%80%D0%B0%D0%B1%D0%B0%D1%82%D1%8B%D0%B2%D0%B0%D0%B5%D1%82%20%22Center%22.%20%D0%94%D0%B0%D0%BD%D0%BD%D1%8B%D0%B5%20%D1%84%D1%83%D0%BD%D0%BA%D1%86%D0%B8%D0%B8%20%D0%BD%D0%B0%D1%85%D0%BE%D0%B4%D1%8F%D1%82%D1%81%D1%8F%20%D0%B2%D0%BE%20%D0%B2%D0%BA%D0%BB%D0%B0%D0%B4%D0%BA%D0%B5%20%22shortcuts%22.&lang=ru-enПеревести вGoogleBingI do not speak English I write through the translator, sorry. I came out writing on 2 teams. When the clamping function is activated "Zero While held", when released triggered "Center". These functions are located in the tab "shortcuts".I do not speak English I write through the translator, sorry. I came out writing on 2 teams. When the clamping function is activated "Zero While held", when released triggered "Center". These functions are located in the tab "shortcuts".
            
                
            
                
            
                
            
                
            
                
            
        https://translate.yandex.ru/?text=%D0%AF%20%D0%BD%D0%B5%20%D0%B3%D0%BE%D0%B2%D0%BE%D1%80%D1%8E%20%D0%BF%D0%BE%20%D0%B0%D0%BD%D0%B3%D0%BB%D0%B8%D0%B9%D1%81%D0%BA%D0%B8%20%D0%BF%D0%B8%D1%88%D1%83%20%D1%87%D0%B5%D1%80%D0%B5%D0%B7%20%D0%BF%D0%B5%D1%80%D0%B5%D0%B2%D0%BE%D0%B4%D1%87%D0%B8%D0%BA%2C%20%D0%B8%D0%B7%D0%B2%D0%B8%D0%BD%D0%B8%D1%82%D0%B5.%20%D0%AF%20%D0%B2%D1%8B%D1%88%D0%B5%D0%BB%20%D0%B8%D0%B7%20%D0%BF%D0%BE%D0%BB%D0%BE%D0%B6%D0%B5%D0%BD%D0%B8%D1%8F%20%D0%B7%D0%B0%D0%BF%D0%B8%D1%81%D0%B0%D0%B2%20%D0%BD%D0%B0%20%D0%BE%D0%B4%D0%BD%D1%83%20%D0%BA%D0%BD%D0%BE%D0%BF%D0%BA%D1%83%202%20%D0%BA%D0%BE%D0%BC%D0%B0%D0%BD%D0%B4%D1%8B.%20%D0%9F%D1%80%D0%B8%20%D0%B5%D1%91%20%D0%B7%D0%B0%D0%B6%D0%B0%D1%82%D0%B8%D0%B8%20%D1%81%D1%80%D0%B0%D0%B1%D0%B0%D1%82%D1%8B%D0%B2%D0%B0%D0%B5%D1%82%20%D1%84%D1%83%D0%BD%D0%BA%D1%86%D0%B8%D1%8F%20%20%22Zero%20While%20held%22%2C%20%D0%BF%D1%80%D0%B8%20%D0%BE%D1%82%D0%BF%D1%83%D1%81%D0%BA%D0%B0%D0%BD%D0%B8%D0%B8%20%D1%81%D1%80%D0%B0%D0%B1%D0%B0%D1%82%D1%8B%D0%B2%D0%B0%D0%B5%D1%82%20%22Center%22.%20%D0%94%D0%B0%D0%BD%D0%BD%D1%8B%D0%B5%20%D1%84%D1%83%D0%BD%D0%BA%D1%86%D0%B8%D0%B8%20%D0%BD%D0%B0%D1%85%D0%BE%D0%B4%D1%8F%D1%82%D1%81%D1%8F%20%D0%B2%D0%BE%20%D0%B2%D0%BA%D0%BB%D0%B0%D0%B4%D0%BA%D0%B5%20%22shortcuts%22.&lang=ru-enПеревести вGoogleBingI do not speak English I write through the translator, sorry. I came out writing on 2 teams. When the clamping function is activated "Zero While held", when released triggered "Center". These functions are located in the tab "shortcuts".I do not speak English I write through the translator, sorry. I came out writing on 2 teams. When the clamping function is activated "Zero While held", when released triggered "Center". These functions are located in the tab "shortcuts".

sthalik, 2016-10-23:
Hey,

Is it working for you now?

sh

BECTHuK, 2016-10-23:
Yes, now it works better. Thank you!

sthalik, 2016-10-23:
The keybindings aren't documented.

sthalik, 2016-11-11:
@BECTHuK a few are. There's a description before the actual list of keybindings. Is that enough?

closed 2017-02-27 by sthalik