opentrack/opentrack #483: freetrack 2.0 Enhanced refuses to work with Elite: Dangerous

amadeus1171, 2016-10-26:
It was working one day and the next, nothing.

I'm using EDtracker as the joystick input with the correct axes mapped.
Output is freetrack 2.0 Enhanced.

I've tried uninstalling/reinstalling.
I've tried choosing from the LocateDLL option, both the 64 and 32 bit versions.
I've tried all the variations of Use/Hide FreeTrack/TrackIR
I've tried changing permissions.
I've tried turning off the MCaffee Virus checker and firewall.
I've tried opening up ports.
I've tried running as administrator.
I've tried the calibrated elbow.

Nothing seems to fix the problem.

I've run freetracktest.exe, and it does show that it's receiving vectors.  But Elite: Dangerous simply does not want to work with it anymore.  It's driving me nuts!  Other than the above mentioned, I hadn't changed a thing before it mysteriously stopped working.

What am I doing wrong?

MathijsG, 2016-10-26:
Do any of the builds here: https://github.com/opentrack/opentrack/releases work or won't they all?

sthalik, 2016-10-26:
Install opentrack on the same partition as ED.

amadeus1171, 2016-10-26:
I only have one partition for the OS and everything else.  I will try previous builds.  But as I said, it worked previously but then suddenly stopped.  And I have tried previous builds at one point or another as they're released.

sthalik, 2016-10-26:
Use the installer in case you used the archive.

amadeus1171, 2016-10-26:
I used the installer.
I didn't use a previous version.  I remembered something that I did before that made it work.  I created an empty profile and didn't modify it, and it worked!  I then switched back to my Elite: Dangerous profile and it continued to work.  What the heck?!

amadeus1171, 2016-10-29:
I'm going to include two profiles.  One is test.ini and the other is elite dangerous.ini

test.ini works when I start Elite: Dangerous with opentrack already running.
elite dangerous.ini does not work with opentrack already running, not running, w/e.

I have to start Elite: Dangerous with the test.ini profile, and once Elite: Dangerous starts, I switch over to the elite dangerous.ini profile and it continues to work.  Can anyone figure out what's going on here?

test.ini
`[migrations]
last-migration-at=20160917_00~

[modules]
tracker-dll=Joystick input

[tracker-joystick]
axis-map-1=0
axis-map-2=0
axis-map-3=0
axis-map-4=2
axis-map-5=3
axis-map-6=4
joy-guid={ACC24E70-A80B-11E5-8002-444553540000}

[opentrack-ui]
center-at-startup=true
compensate-translation=true
use-camera-offset-from-centering=false
x-source-index=0
y-source-index=1
z-source-index=2
`

elite dangerous.ini
`[Curves-tx]
point-count=1
point-0-x=100
point-0-y=45

[HT-Tracker]
camera-name=Logitech HD Pro Webcam C910
fov=56
fps=0
resolution=1

[keyboard-shortcuts]
keycode-center=Alt+F12
keycode-toggle=Ctrl+F12
keycode-zero=Shift+F12

[opentrack-ui]
camera-pitch=0
camera-roll=0
camera-yaw=0
center-at-startup=false
compensate-translation=false
compensate-translation-disable-z-axis=false
filter-dll=Accela
pitch-alt-axis-sign=false
pitch-invert-sign=false
pitch-source-index=4
pitch-zero-pos=0
protocol-dll=freetrack 2.0 Enhanced
roll-alt-axis-sign=false
roll-invert-sign=false
roll-source-index=5
roll-zero-pos=0
tracker-dll=Joystick input
use-system-tray=false
x-alt-axis-sign=false
x-invert-sign=false
[... 173 lines cut ...]

[spline-Z]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@>\0\0\0\0\0\0@>\0\0\0\0\0\0)

[alt-spline-Z]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@>\0\0\0\0\0\0@R\xc0\0\0\0\0\0)

[spline-yaw]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x3@(\0\0\0\0\0\0@[\xc0\0\0\0\0\0@.\0\0\0\0\0\0@d\0\0\0\0\0\0@1\0\0\0\0\0\0@f\x80\0\0\0\0\0)

[alt-spline-yaw]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@7\xc4\n\x94\xb8\x94#@f\x80\0\0\0\0\0)

[spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2@*\0\0\0\0\0\0@U\x80\0\0\0\0\0@0\0\0\0\0\0\0@f`\0\0\0\0\0)

[alt-spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@f\x80\0\0\0\0\0@f\x80\0\0\0\0\0)

[spline-roll]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@Y\0\0\0\0\0\0@f\x80\0\0\0\0\0)

[alt-spline-roll]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@f\x80\0\0\0\0\0@f\x80\0\0\0\0\0)
`

sthalik, 2016-10-30:
`used-interfaces=1`

Either 2 or 0

amadeus1171, 2016-10-30:
What?  Opentrack configured this.

sthalik, 2016-10-30:
This isn't the default value. As you can see, empty config defaults to a value of 0.

amadeus1171, 2016-10-30:
My question, though, is why did Opentrack configure an invalid value?

sthalik, 2016-10-30:
You selected it in freetrack protocol options -- protocol visibility list.

amadeus1171, 2016-10-31:
...
...
...
Why does it offer an invalid value in the first place?  Forgive me, but that doesn't seem to make any sense.

sthalik, 2016-10-31:
It makes sense for other games.

closed 2016-11-03 by sthalik