opentrack/opentrack #484: 180°+180° - Reverseview available?

vinKaiZen, 2016-10-27:
At the moment i´m using opentrack with a self created FreePIE sript to realize a 180° Front & 180° Reverse view. (I´m just pushing a button and the script adds +180° to the send TrackIR coordinates)

Is there a possibility to integrate such feature in Opentrack?

Second idea Same feature without any push button:
After passing the "180° Frontview Area" (@-90° +90°) there will be added 180° to the send TrackIR Coordinates automatially. If you want to go back to the "Frontview" just pass borders of the Reverseview Area (@-90° +90°) and there will be substracted 180° to the send TrackIR Coordinates.

I hope i get understood :)

sthalik, 2016-10-27:
What's meant to happen once you reach +-90?

vinKaiZen, 2016-10-27:
if passed the ±90° or ±110° viewswitching borders (like the picture above, blue marked) the "used" view Area will switch to the other side (green marked view area). 

Example. If you are sitting in a traktor and want to drive backwards to park a trailer, you turn your head more than ±90° (or 110°) degrees and the view area switch to the backward view.
Then move the head normal into the moitor you an use the "backward view" like normal front view".
If finished parking, turn your head once aggain more than the ±90° or 110° "viewswitching borders" to switch the view area back to the frontview.

sthalik, 2016-10-28:
Does it mean that with it activating at +-90, then add to the yaw +-180 with the correct sign? You want moving leftward suddenly jump rightward once the threshold is crossed?

vinKaiZen, 2016-10-28:
Thats what i mean.
That the hard cut version ^^
Maybe the switch between the front and reverse view could be done smootly, like a wipe from one desktop to another on a tablet or smart phone :)

sthalik, 2016-10-28:
How is this more natural? I can hardly imagine drivers sitting backward in the chair while looking into their rear window.

vinKaiZen, 2016-10-28:
Thats exactly the actual situation. I dont want to turn my head 180° while my monitor is in front.

I just want to turn it 90° (the borders) and then back to middle to switch the ingame view from the front to the the rear window. Same procedure from the rear to the front.

If that is not doable, please add a button which adds 180° to the actual view

sthalik, 2016-10-28:
You'd want turning back to origin have the view remain backward? I won't implement that.

Unless using a VR helmet, I recommend you increase rotation sensitivity so 90 degrees yaw need only 15-25 of the user's yaw.

vinKaiZen, 2016-10-28:
That is exactly the situation i don´t want to have. (see my first post, first 3 lines)
I want a 1:1 rotation sensivity. 45 head = 45° view change.
Thats why i need a button which can switch my view to behind with only 1 button if pressed

vinKaiZen, 2016-10-28:
I made a short video of my freepie script with the function i want to have native in opentrack.

sthalik, 2016-11-11:
Please comment on #492

closed 2017-02-21 by sthalik