opentrack/opentrack #487: Request +-180 degree normalisation as an option.

H-Bear-22, 2016-11-03:
G'day @sthalik 

I was wondering if the "Trackers submitting rotation over +-180 will have it normalized" mentioned in the release points for rc13 could be made a user-selected option?

I have checked the output csv and found that my joystick input device provides a raw maximum input of 180.003 which is 'normalized' to -1

85.635, effectively switching my view from full right to full left when my head is turned fully to the right.

I just have a 'normal' setup with one monitor in front of me.

Thanks for your time,

 - Bear

[test.zip](https://github.com/opentrack/opentrack/files/568214/test.zip)

sthalik, 2016-11-03:
This is a rounding bug.

Please see if the joystick build fixes. https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

sthalik, 2016-11-03:
In this build

- joystick is clamped to -180->180
- overall, trackers need to be over +-180.01 to break

sthalik, 2016-11-03:
There's a rounding error in dinput code. Fixed.

sthalik, 2016-11-03:
I confirmed it works with my own joystick.

closed 2016-11-03 by sthalik

H-Bear-22, 2016-11-03:
Perfect!

Thank you very much for your work @sthalik 

 - Bear

sthalik, 2016-11-03:
Thanks for confirmation.

commit e26830bf (2016-11-03): logic/tracker, dinput, tracker/joystick: fix rounding errors

Issue: #487
Reported-by: @H-Bear-22
Closes #487

Also use a power of two for joy_axis_size. Floats are base 2.

axis_max in tracker/joystick had a fencepost error.