opentrack/opentrack #489: Gyroscope turns instantly from 180 to -180

ZioPao, 2016-11-03:
I was trying the latest version of opentrack with my Nexus 5, but I couldn't make it work since it was costantly turning back from 180° to -180°, and it makes it pretty much unususable. 
I tried some really old version of this (like r19) and everything works fine, but with some of the latest ones it just doesn't.
Sorry for my english, is it a bug or what?

sthalik, 2016-11-03:
Was 180 the correct value?

ZioPao, 2016-11-03:
yes.. But obviously the -180 was wrong since it flips completely. Also, for some reasons it seems like it confuses the rolls with the yaw.

sthalik, 2016-11-03:
It depends how you have the phone rotated. Go to axis remap in the options window.

ZioPao, 2016-11-03:
no... I'll upload a video tomorrow, it's not like that. I tried to remap it, and in the old version it works fine. I have these problems only in the newer versions, it has nothing to do with the orientation of the phone, or at least that's not what it looks like.

sthalik, 2016-11-04:
In tracker options adjust relevant axis by -180.

sthalik, 2016-11-04:
That is, for freepie tracker. Other trackers don't have it.

ZioPao, 2016-11-04:
Ok so I just tried that but obviously it wasn't really a fix since some times I have to reset it since it isn't really precise and it would have been stuck at -180 after it. So I just reset it and now it doesn't seem to have any problems... I literally changed the Yaw to -180 and then reverted it to 0, and bam, it fixed even the Roll\Pitch problem. Strange...

sthalik, 2016-11-04:
Any magnets, like headphones or speakers?

ZioPao, 2016-11-04:
I don't think so, but as I said I tried the older version after I saw the the new one didn't work, and it worked fine without a single problem.
Anyway I think this can be closed since it seems something pretty rare as I'm seeing, and this "set to -180 and back to 0" seems to have fixed everything for me

sthalik, 2016-11-04:
Do you need to do this also on a config made from scratch?

ZioPao, 2016-11-04:
I just tried again, making these changes from a newly made config, and for some reasons... it doesn't work anymore. Still, the old version even after this try still works fine.

sthalik, 2016-11-11:
Use the "add angle value" option for the tracker to put each rotation axis as close to zero as possible. You'll see it take effect on "raw value" display. This is likely caused by the gimbal lock multiple solutions problem. That way I can be sure it's solvable using the same trick as used in the Rift tracker.

vikingsail, 2016-11-16:
Had some weird problem while the yaw axis surpasses about 70 degrees, it randomly jumps all the way to 180. it doesn't always happen, but once it happened, there's no way to fix it unless you restart the tracking procedure. 
sthalik provided the solution which worked for me - at least for now - by adding -180 to the yaw axis in the freebie tracker settings.

Phil8811, 2016-11-22:
Hi, I have the same problem, after few time, on stm32f3 card in USB with input set as oculus 0.2.5 the pitch movement become pitch +roll move in the game for it come back normal I need to stop tracking and restart tracking
PS: which one previous version was working without this bug please? 
 
Best regards

ZioPao, 2016-11-22:
I'm using 2.3 rc19p3 and it works fine, no problems at all.

ZioPao, 2016-11-22:
I tried newer opentracks with both my nexus 4 and nexus 5, neither of them work as they should. I doubt it's a device related problem since older versions fixed every single problem I had with it

Phil8811, 2016-11-22:
Thx ZioPao,  I can't try with this version because oculus setting don't work with my card in this version. 
Vikingsail , I have same feeling than ZioPao, because if the Gyroscop was the problem stop and restart tracking mustn't fix the problem but for few time (before it start again) it fix it.
I mean before the bug start pitch is pitch and roll is roll, after bug start pitch and roll seem like mixed together, if you stop tracking and start tracking pitch comeback as pitch only and same for roll I suppose if it was gyroscop problem the axis must be mixed in spite of stop/start don't you think? 

Best regards

sthalik, 2016-11-29:
What are the typical values where pitch and roll get confused together?

sthalik, 2016-11-29:
Also do you get it in raw values or in game values only?

Phil8811, 2017-01-18:
Hi when the bug appear when you move physically in pitch only in Opentrack
in raw value the pitch and roll get about same value for both axis (about
yaw axis I'm not sure if the bug affect this axis too)

Best regards

Phil8811, 2017-01-18:
It's not easy to describe the bug because value change all the time

Ok till now I mapped my center button on "Center" in Option/Shortcuts but I
just discovered I can use "restart tracking"
Need test more maybe it could help but the best should be remove the bug if
possible

PS: do you think change the 2d image of Octopus for a 3d object? Should be
great

Best Regards

sthalik, 2017-01-18:
@Phil8811 it woulda been fun to have a 3D octopus but it's too much work for what it's worth.

sthalik, 2017-01-18:
If there's a discontinuity with rotation we'd need that matrix -> rotation vector -> matrix code that Vive also needs.

sthalik, 2017-02-27:
Possibly related to #554

sthalik, 2018-05-14:
Should be fixed/changed in dev build.

closed 2018-05-14 by sthalik