opentrack/opentrack #490: Using mapping profiels from older releases

IvanDolvich, 2016-11-08:
For a long time I was using Opentrack 2.3 rc21p11. One reason for doing that and not using a later release was that I could not use my mappings for the axis from my older version. The profile is shown and can be selected but it has no influence on the mapping properties. They are still the preset defaults. Did anyone run into the same problems and can help me solving this?

Len62, 2016-11-08:
I thought that RC12 implemented migration of settings, however, I think the mapping has been updated since RC12, so maybe that broke the migration?

IvanDolvich, 2016-11-08:
The settings for camera resolution, camera offset etc. are ported to the new version...

Len62, 2016-11-08:
Right, but possibly, due the changes in the mapping, the migration does not work on the mappings. I'm sure @sthalik  will chime in soon.

sthalik, 2016-11-09:
Remove the rc11 mapping sections before updating to rc12. Otherwise rc12 won't migrate cause it'll consider new mappings to exists.

sthalik, 2016-11-09:
I'll add a warning to the rc11 page.

sthalik, 2016-11-09:
That is -- the new sections are binary data as `Variant`. You'll find six of them.

closed 2016-11-09 by sthalik