opentrack/opentrack #492: support for conditional specialized behavior

sthalik, 2016-11-11:
There are now a few requests for specialized behavior that's hard to implement using a user interface. Some of these are

- reduce relative translation to allow for precise gun aiming (private message via forum.il2sturmovik.ru
- nonlinear mapping scaling -- #423
- particular reverse-view behavior -- #484

So how about Python scripting? Any user interface to enter logical expressions and the result would be way more cumbersome than a script with a few constants that can be mostly copy-pasted.

- Use a Python decorator to specify input data requested and what's modified.
- Use an arbitrary ipython-like syntax highlight.
- Save scripts as plain text files in the configuration directory's subdirectory, say, `scripts/`.
- Provide docs on functionality "recipes".
- Strongly state that no expert knowledge is really necessary to use the scripting functionality.
- Implement a FFI type for `XYZ` `ypr` vector that needn't allocate each tick.
- Allow script injection between each pipeline stage -- `logic/tracker.cpp`.

vinKaiZen, 2016-11-12:
Sounds fantastic.
To: [#484](https://github.com/opentrack/opentrack/issues/484): This is my working phyton script for EDTracker=>vJoy=>Opentrack

[Opentrack - Phython Reverse View Script.zip](https://github.com/opentrack/opentrack/files/587490/Opentrack.-.Phython.Reverse.View.Script.zip)

sthalik, 2016-11-29:
I'm slowly working on this stuff. There's a straightforward way to write most of the support code in pure Python and avoid significant performance loss in the tracking loop.

We should also get the option to programatically define keybindings, as well as have declarative user interface API in Python without any dependencies on PyQt or similar.

I'm writing this to remember.

closed 2018-05-14 by sthalik