opentrack/opentrack #493: [X-Plane] OpenTrack and dual monitors

Denco96, 2016-11-12:
I'm using dual monitors setup for X-Plane and I've noticed that when using TrackIR view doesn't move in sync. One monitor moves faster that the other. The issue is not present if I use my move to view around the cockpit. I'm using Delan Clip with Open Tracker Input.

Can anyone lend any advice on this problem?

Denco96, 2016-11-12:
I think this happens because of the FOV distortion. Will report back once I find out more.

sthalik, 2016-11-12:
Are you using PilotView? You should.

Denco96, 2016-11-12:
I'll give it a go. Otherwise I'm using X-Camera.

sthalik, 2016-12-24:
How is it?

Denco96, 2016-12-24:
After some searching online I think I've discovered a possible reason why this is happening and it's entirely unrelated to Open Track. Because the monitors are run using the extended mode, instead of the surround mode, it is possible that the fps are not the same on both monitors which could cause the anomalies I'm experiencing.

sthalik, 2016-12-25:
I hope it works for you in the end. I'm closing the issue since it's unrelated to opentrack.

closed 2016-12-25 by sthalik