opentrack/opentrack #494: Output to SteamVR?

MaVe-64, 2016-11-13:
Can't find this anywhere, is it possible to have Opentrack output to SteamVR protocol?

If someone knows how, please say! Thanks!

sthalik, 2016-11-13:
Duplicate of #352

Prestoned, 2017-03-02:
no its not its the opposite opentrack>steamvr service not SteamVR>TrackIR

sthalik, 2017-03-03:
@Prestoned Is there a documented way to do this?

3DCont, 2017-05-21:
This topic is different from #352 , please reopen it.
It would be very useful for who use moonlight with google cardboard, or for anyone with an aio hmd with hdmi input.

It seems to be possible through the steamvr null hmd driver.
https://github.com/ValveSoftware/openvr/blob/master/samples/driver_sample/driver_sample.cpp

There are discussions about the null driver receiving UDP.
https://groups.google.com/forum/#!topic/psmoveservice/nFB1po4OmLU
https://github.com/HipsterSloth/PSMoveSteamVRBridge/issues/1

------------

Wishlist:
- SteamVR HMD driver with suport to opentrack.
- Opentrack with secondary input. (FreepieUDP to receive gyro data from phone and pointTrack to positional tracking).
- Single Light Ball positional tracking.(Z axis based on size)

sthalik, 2017-05-25:
Thank you for the information. I'll look into it.

sthalik, 2017-05-25:
We should be able to create another driver, then pass data from it, except for orientation to supply on our own.

It's just that it's gonna take a real while before I start this particular project. I'm knee-deep in other matters and there are many items queued up for opentrack already.

r57zone, 2017-05-25:
This is a very important feature, as there are already many games and applications with OpenVR support.

It seems [here](https://github.com/ValveSoftware/openvr/blob/master/samples/driver_sample/driver_sample.cpp#L348) is necessary to give data quaternion, [which can be calculated](https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles) from the yaw pitch roll.

```
static Quaterniond toQuaternion(double pitch, double roll, double yaw)
{
	Quaterniond q;
	double t0 = std::cos(yaw * 0.5);
	double t1 = std::sin(yaw * 0.5);
	double t2 = std::cos(roll * 0.5);
	double t3 = std::sin(roll * 0.5);
	double t4 = std::cos(pitch * 0.5);
	double t5 = std::sin(pitch * 0.5);

	q.w() = t0 * t2 * t4 + t1 * t3 * t5;
	q.x() = t0 * t3 * t4 - t1 * t2 * t5;
	q.y() = t0 * t2 * t5 + t1 * t3 * t4;
	q.z() = t1 * t2 * t4 - t0 * t3 * t5;
	return q;
}
```

[I tried to change](https://github.com/HipsterSloth/PSMoveSteamVRBridge/issues/1#issuecomment-291876635) the default values of the quaternion in "pose.qDriverFromHeadRotation", but I could not get the driver to work ("driver_sample"). In "steamvr.vrsettings" I replaced id, driver_name and so on.

The "driver_null" driver in OpenVR [works without problems](https://imgur.com/a/0wTjb), but there is no tracking. Perhaps "driver_sample" is not a "driver_null" in git openvr  and it needs to add code.

sthalik, 2017-05-26:
See if it's loaded at all. Use "sysinternals procmon" for that. It should be opened by "vrserver.exe". "driver_null" looks like a non-driver. "driver_sample" is a driver in its own right, just like the one for htc vive, etc.

r57zone, 2017-05-26:
Procmon shows that load image from "driver_null".
[image]

sthalik, 2017-05-26:
Does "dependency walker" list any missing libraries? I recommend using static multithreaded CRT rather than the dynamic one.

It should load from "driver_sample". Furthermore, it'll load the first working driver in the list, not all of them. Remove the working ones above "driver_sample". Loading an image isn't enough, it needs to be loaded as a module.

r57zone, 2017-06-08:
The sample driver is the same as the null driver[[1]](https://github.com/ValveSoftware/openvr/issues/546).  Maybe specify exactly the same constants id, name or etc? 
It also looks like need to replace the original Null driver[[2]](https://github.com/ValveSoftware/openvr/issues/97).

sthalik, 2017-06-08:
What about the display data? I have a feeling games are gonna stop rendering once they detect a SteamVR driver.

r57zone, 2017-06-09:
The games & dashboard [work well](https://imgur.com/a/0wTjb).

LonnyEnds, 2017-06-09:
Eagerly watching this as in same situation as Issue #633. Take as long as needed this would be huge as lack of 6DOF and bad drift are killers to the Pimax value. Thank you all for working on this!

sthalik, 2017-06-09:
But this isn't related only to head tracking, but VR support for third-party devices. Am I not correct? VR support is outside the scope of the opentrack project.

r57zone, 2017-06-09:
Third-party devices can be connected by a separate driver. I already tested this with [VirtualController](https://github.com/SecondReality/VirtualControllerDriver). Only a driver with head tracking is enough.

r57zone, 2017-06-21:
I was mistaken (not pose.qDriverFromHeadRotation), it is necessary to do so like this - [spvr](https://github.com/burrito82/spvr.driver/blob/808eb2c63b1d4240208398a8ebc6f261ef9706c1/HmdDriver.cpp#L142)(old driver, [previously worked](https://burrito82.github.io/spvr/), now does not work).
> double t0 = cos(yaw * 0.5);
> double t1 = sin(yaw * 0.5);
> double t2 = cos(roll * 0.5);
> double t3 = sin(roll * 0.5);
> double t4 = cos(pitch * 0.5);
> double t5 = sin(pitch * 0.5);
> 
> pose.qRotation.w = t0 * t2 * t4 + t1 * t3 * t5;
> pose.qRotation.x = t0 * t3 * t4 - t1 * t2 * t5;
> pose.qRotation.y = t0 * t2 * t5 + t1 * t3 * t4;
> pose.qRotation.z = t1 * t2 * t4 - t0 * t3 * t5;
In function "virtual DriverPose_t GetPose()" on [driver_sample / driver_null](https://github.com/ValveSoftware/openvr/blob/master/samples/driver_sample/driver_sample.cpp). 
Also need to add the implementation of a UDP server which starting on load dll library (in main), and global variables float: yaw, pitch, roll and the function of reading port 4242 from this UDP server. For example [implementation on Delphi](https://github.com/r57zone/Steering-wheel-for-PC-with-OpenTrack-and-Android/blob/master/Project1.dpr#L315-L342).

sthalik, 2017-06-22:
@r57zone if there's a ready driver I will integrate opentrack support. But writing a VR passthrough driver as part of the opentrack project is overkill.

r57zone, 2017-06-22:
I made a [driver](https://github.com/r57zone/OpenVR-head-tracking-with-buttons) in which can rotate your head using the numpad buttons. It remains only to add reading from OpenTrack.

@sthalik I understand, maybe someone another else will be interested to write this.

sthalik, 2017-06-22:
@r57zone if you listened on an UDP socket, opentrack spews out 6 `double` values. Respectively, XYZ position, then yaw, pitch, and roll in degrees (not radians). Is this feasible for you?

sthalik, 2017-06-22:
Also we have a TrackIR client to connect via the API without signing any NDAs.

r57zone, 2017-06-22:
Yes, I know, I try to do.

sthalik, 2017-06-22:
If you need help with the TrackIR client just ask. There's a preprocessor definition `MY_DEMO` and it should show the usage pattern.

r57zone, 2017-06-24:
I did it, but I have an input lag, need to change WinSock reading to another separate thread. If someone can help, here are the [repository](https://github.com/r57zone/OpenVR-OpenTrack) (I don't know C \ C++).

Also here is the [implementation of reading the UDP over network](https://github.com/r57zone/OpenTrack-UDP-receive) on WinSock (Windows) and C, if someone else will need it someday.

[binary_null_driver_with_config](https://github.com/opentrack/opentrack/files/1099830/binary_null_driver_with_config.zip)

LonnyEnds, 2017-08-04:
So I tried r57zones method and I just wanted to confirm that this will only work with screen mirroring/ extended mode vs Direct mode under steam VR. I have a device that will run steam VR in direct mode but want to inject this tacking data from UPD? Is that possible?

r57zone, 2017-08-14:
Updated [OpenVR driver](https://github.com/r57zone/OpenVR-OpenTrack). Fully support extended mode.

r57zone, 2017-09-07:
Added positional tracking, but testing is necessary. It would be cool if someone could check with Aruco Paper Marker or another tracker.

sthalik, 2017-09-07:
@r57zone, can you check against a prerecorded video with ManyCam? I have a few.

r57zone, 2017-09-07:
@sthalik yes, could you upload them somewhere?

sthalik, 2017-09-07:
Send me an email please. The address is on my profile page.

r57zone, 2017-09-19:
As controllers, you can use paper trackers (with Aruco paper marker), made as [Zappar paper controller](https://i.imgur.com/AhO2GV7.jpg). Sending button presses can be done using Arduino.

r57zone, 2017-10-18:
[Driver](https://github.com/r57zone/OpenVR-OpenTrack) updated to latest SDK 1.0.10. I also created a standard for DIY and other VR devices. Сan be combine OpenTrack trackers and others for proxy VR drivers or etc.

sthalik, 2017-10-19:
Hey,

Did you receive the Aruco videos via email?

Also were you able to integrate data through something other than the UDP protocol?

r57zone, 2017-10-19:
@sthalik No, I did not receive it, I sent email to you 07/09/2017. In the spam folder, too.

Yes, I will make another driver that will support FreeTrack 2.0.

sthalik, 2017-10-19:
I sent you a reply on 2017-09-08, 07:02. If you still don't have it, add "sthalik" on Skype or Steam and you'll get it.

r57zone, 2017-10-19:
I made a [driver with FreeTrack support](https://github.com/r57zone/OpenVR-OpenTrack), however, in order for everything to work, also need a 64 bit version of the freetrackclient.dll module. With the 32-bit version, SteamVR will only work on 32-bit systems. It seems that OpenTrack needs a 64-bit version of the freetrackclient module.
I made a branch in the registry ``HKEY_CURRENT_USER\Software\OpenTrack`` with the ``FreeTrack`` and ``FreeTrack64`` values with paths to these libraries.

@sthalik ok, when I get ready to test positional tracking, I will contact you.

sthalik, 2017-10-19:
@r57zone you don't need to use it. Implement the API to read the mapping directly.

sthalik, 2017-10-19:
Alternatively I can provide a 64-bit version.

r57zone, 2017-10-19:
I made a mistake and used the wrong API. Yes, it would be nice to get a 64 bit version.

sthalik, 2017-10-19:
Okay. Remember that there are no multiple calling conventions on amd64 so drop `stdcall`.

[foo.zip](https://github.com/opentrack/opentrack/files/1400050/foo.zip)

sthalik, 2017-10-19:
Tell me if it's broken or anything.

r57zone, 2017-10-20:
Works, thank you. The driver is ready.

sthalik, 2017-10-20:
I can point to your driver in opentrack's readme and stable release news. Send a URL if you want it.

r57zone, 2017-10-20:
It would be great - "https://github.com/r57zone/OpenVR-OpenTrack"
Thank you.

r57zone, 2017-10-26:
Added distortion lenses. Many people failed to adjust manually, so I also made a program for a simple setup of SteamVR (3 click).

sthalik, 2017-10-28:
@r57zone, you can take the freetrack library source and use it directly, rather than as separate dll. It's a bit more robust.

You're now referenced from the project's readme. Feel free to create a wiki page on the opentrack project. I'm unable to provide any support myself.

r57zone, 2017-11-01:
Added direct read FreeTrack.

gb2111, 2017-11-08:
hi. i work on alternative solution.
it allows you make head tracking to most of headsets like pimax4k, oculus dk2 but also Android with RiftCat.
I can share link if you don't mind that it is chargeable.

Greg

r57zone, 2017-11-08:
@gb2111 Drivers for HMD are already many, but there are no drivers for cheap controllers. It would be interesting to see such a driver. For example, web cams and paper controllers with aruco markers.

sthalik, 2017-11-09:
@gb2111 you're allowed to share links to commercial projects here in this discussion thread.

sthalik, 2017-11-09:
@r57zone you can do it already. Since you receive data via the freetrack protocol, select Aruco as input and it'll work.

gb2111, 2017-11-09:
Hi,
Here is driver a test version of freetrack support:
https://drive.google.com/file/d/1QPhoDIwuSedRh8JlCNQkPyM725uJScqD/view?usp=sharing

You need create account on my site www.driver4vr.com. Then tell me login here or contact form and I will gratnt full access.
Then run SteamVR. You will see Driver4VR UI starting and after login all you need is to select FreeTrack as source and press Start. Attach screen.
[image]

In Opentrack you can use any method and as output put Freetrack and voila.
I assume you have your headset either DK2, Pimax, Deepoon, or via Android or iOS.
This should add ability to see position to headset immediately even if you simply glue aruco marker to front of your headset :)

If you open Settings to the right you will see extra option to include rotation. This one will probably need more testing to make it alighed with head orientation. But basically we can bypass yaw info from headset gyro and remove drift. 

Let me know if you ahve any questions.
Greg

sthalik, 2017-11-09:
@gb2111 interesting. I can't personally test this due to lack of any HMD.

Remember to mount an aruco marker at an angle as per wiki. You'll save yourself problems.

gb2111, 2017-11-09:
@sthalik , thanks for info. I am going to make a video where I use it with some simulators from Steam. Like Warthunder or Project Cars. If that will give me positional within limits of sitting game then it will be really great. Only problem might be small FPS. I was testing with leds and I got like 8FPS which is not too much... Something like 30 would be much better but don't know if this is possible.

ps. I was not aware they sell Mocne any more. Lucky I don't smoke any more :)

sthalik, 2017-11-09:
This is a stock photo.

I smoke all kinds of crap and it's no pleasure at any time. Still it's better than getting addicted to alcohol or hard drugs.

sthalik, 2018-08-04:
If there's something that can be cleanly integrated send a pull request.

closed 2018-08-04 by sthalik

r57zone, 2018-08-17:
I made a standard for DIY VR - [True OpenVR](https://github.com/TrueOpenVR), with SteamVR driver. We can use various trackers for both HMD and controllers. We can make controllers on Arduino, from Android smartphones and etc.

r57zone, 2020-03-01:
I updated the [OpenTrack SteamVR bridge driver](https://github.com/r57zone/OpenVR-OpenTrack) to the latest SDK and now it is easier to run, just unzip it to a folder and done.

r57zone, 2022-11-01:
Driver [updated](https://github.com/r57zone/OpenVR-OpenTrack), crouch button added.