opentrack/opentrack #499: Star Citizen, Avegant Glyph

Bunitj, 2016-11-22:
Hello

I have just started playing around with [OpenTrack](https://github.com/opentrack/opentrack/releases) in an effort to use my [Avegant Glyph](https://www.avegant.com/) in [Star Citizen](https://robertsspaceindustries.com/).  Neither is set up for the process, but I am trying to work around it as someone who does not know anything about coding or this area.  Just someone who is interested in solving the problem.

Here is the thread on the glyph Forums that I have been trying to solve this.
https://www.avegant.com/forums/discussion/3919/youtube#Item_1

and here is the thread on the Star Citizen forums.
https://forums.robertsspaceindustries.com/discussion/329427/avegant-glyph#latest

And this has led me to here.  To save you reading through the threads it has come down to I finally worked out how to edit the XML files so that the game is finally, sort of, accepting information from the head set, but the information is very... I think untidy is the best word I can think of it.  

It was at the point when I started looking for 3rd party software that I found something un expected with the headset.  I knew it had head tracking capabilities, and a switch to turn it on and off, as was published.  But there are 2 different modes.  On and Off is not actually on and off.  When head tracking is turned "On" the headset (which is connected by USB as well as HDMI, it allows you to use the headset to control the mouse.  What I discovered when I put in 3rd party software is that when head tracking is "Off" it is still sending data and is recognised by my PC as a joystick.  This, as far as I can tell, is not a listed function, though a function I am trying to use!

https://www.youtube.com/watch?v=Jr_VCFqgtXQ 
This video shows the tracking data from software called [SCJMapper](https://github.com/SCToolsfactory/SCJMapper-V2/releases) that you can use to create an XML file that will be recognised and map to the game.  The video starts with head tracking "Off", and you can see the data in the [SCJMapper](https://github.com/SCToolsfactory/SCJMapper-V2/releases) window being read, and part way through I turn the head tracking "On" and it takes over the mouse.
Though it does little to refine the data as it comes in.  You can see what I call Dirty Data.  As a result, I get definite response from the headset in the game, but they are unmanageable,  Mostly the camera rotates directly up and sits there.  Rotating the headset in a rolling manner makes the camera look left and right for some reason.  

So at this point, I came across [OpenTrack](https://github.com/opentrack/opentrack/releases).  [OpenTrack](https://github.com/opentrack/opentrack/releases) produced data that I found more useable because it's data point used 360 degrees as its measurement system, where [SCJMapper](https://github.com/SCToolsfactory/SCJMapper-V2/releases) seemed to have infinity. 

As you can see from this video, the data was still a little unrefined in [OpenTrack](https://github.com/opentrack/opentrack/releases).
https://www.youtube.com/watch?v=D1-vjch0SHk

Once I learned a little about the navigation and operation of  [OpenTrack](https://github.com/opentrack/opentrack/releases) I then edited some of the input and out settings, as in the images, below, and the data became what I consider.... Clean.

https://www.dropbox.com/s/2jh2npcwsmhzito/Output%20map%201.JPG?dl=0

https://www.dropbox.com/s/b2i70k6snwj6ign/Output%20map%202.JPG?dl=0

These changed resulted in this data output.
https://youtu.be/tVnJgi6NQuU

[Here is the track Data](https://www.dropbox.com/s/icbxdi7onc1xc6c/Track2.csv?dl=0)
[Here is the .ini file](https://www.dropbox.com/s/m36ohizhrlhmnyu/default.ini?dl=0)

Now I am happy with this output data, but the problem is I have no idea how to use this data in Star Citizen, and this is where I was hoping that someone here could help me.  I am very inexperienced with coding, basically been teaching my-self roughly how to do things since I started out on this project.  There an easy to access [XML file](https://www.dropbox.com/s/xuykrv06j5r02h5/layout_my_joystick.xml?dl=0) that I have become familiar with in the Star Citizen folder that I can use to identify control inputs etc.  I just have no idea how to put this information from OpenTracker, or even OpenTracker its self between the headset and the game.

I am not a programmer, so I have no idea if I am using any of the correct terms of the like.  I hope you understand what I am getting at.

Thanks for your help.
Bunitj

sthalik, 2016-11-22:
Hey,

What protocol does the device send the tracking data over? There are SteamVR, Oculus SDK et al. We also already have support for joystick-like input.

Bunitj, 2016-11-23:
Hello,

Thanks for responding Sthalik.  I am using the joystick input format and it is outputting perfectly from OpenTrack.  The Oculus profiles just give me an error, and I can't see the SteamVR on the list at the moment.  But as far as output is concerned I am happy with with the numbers I am reading.

My issue comes from how do I make the output from OpenTrack be the Input for Star Citizen?

sthalik, 2016-11-24:
Use the "freetrack" output. Use no filter.

Bunitj, 2016-11-29:
Hello @sthalik. I tried that after you suggested it, and it did not seem to help.  I changed the output settings on opentrack and while the changes I made affected the OpenTrack output window appropriately, it had no effect at all within StarCitizen, including such dramatic changes as inverting and changing output target fields.  Any suggestions?

sthalik, 2016-11-29:
From a brief Google search I see this - https://www.reddit.com/r/starcitizen/comments/4kdmlm/psa_to_trackir_users_trackir_for_star_citizen_is/

Unless the game has a way to transfer the pose data (protocol like freetrack, joystick pseudo-device, etc), we can't do anything in this project.

closed 2017-12-02 by sthalik